wokwi/wokwi-features #237: NeoPixel strip, e.g.

alto7, 2021-12-08:
Where is there a list of all available simulator parts? The docs doesn't have it, the in-app dropper doesn't have it, so… 

I have been reduced to mining the *.json from other projects, as it has proven impossible for me to find some of them, or at least digging up a project that has them is faster…

The documentation for every part would be improved if not only the applicable attrs were listed, but also the full json entry such a part would create.

alto777

urish, 2021-12-08:
In short: the docs have most of the parts. 

Some of the parts are still in the prototyping phase, and writing docs does take a considerable amount of type. So I usually wait until I finalize the implementation and then write the docs (e.g. that's the reason why #196 is still open).

> The documentation for every part would be improved if not only the applicable attrs were listed, but also the full json entry such a part would create.

Can you please give an example what you mean here?

alto777, 2021-12-09:
Yes, although I see now it mightn't matter eventually.

If the part is on the pull down, no need to necessarily care. Are the parts in the documents all on the pulldown? Will that continue to be possible (optimist! It'll be unwieldy when wokwi is older…).

If you always provide an example project in the doc page for a part _not_  in the dropdown, one could open that project and take a peek at the *.json, grab the part cut and paste it into the project needing it.

Which I am assuming you must do, as how else do beta parts get into projects? Like I said, I've been finding and cut/pasting various neopixel parts, which I now understand are not done yet.

I have some of these ready to go in a text file. I look forward to messing less with the *.json stuff. :-|

> {
     "id": "neopixels",
      "type": "wokwi-neopixel-canvas",
     "top": 0,
      "left": 300,
      "attrs": {
        "rows": "1",
        "cols": "16",
        "matrixBrightness": "10"
      }
    }

alto777

drf5n, 2022-04-16:
Prior to full documentation, maybe the little tool-tip question-mark could link to the wokwi-feature issue related to it?

The "?" question mark on this LED leads to https://docs.wokwi.com/parts/wokwi-led

[image]

...but the wokwi-neopixel-ring and the wokwi-neopixel-canvas do not have "?" popups and links 

Looking at https://wokwi.com/arduino/libraries/Adafruit_NeoPixel examples, they all use `wokwi-neopixel-canvas`

drf5n, 2022-04-16:
> I have some of these ready to go in a text file. I look forward to messing less with the *.json stuff. :-|

Please post your text file -- pre-formal-documentation it's hard to find out how to use these parts.  Several parts are listed on https://github.com/wokwi/wokwi-features/issues/59 but I can't find examples of all of them.

Logopher, 2022-07-03:
@alto777 Ping. Could you post/link the data you mined? I got here by happening to search "wokwi-neopixel-canvas", luckily for me.

drf5n, 2023-02-03:
Note that the connections for a wokwi-neopixel-canvas are DOUT, VSS, VDD, DIN, and are located on the middle of the bottom edge.

If you want a long strip, It is easiest to plant a small strip in the .json, connect it up, then increase the size. 

```
    {
      "type": "wokwi-neopixel-canvas",
      "id": "neopixels",
      "top": -336.29,
      "left": 145.7,
      "attrs": { "rows": "1", "cols": "16", "matrixBrightness": "10" }
    },
```
and
```
  "connections": [
    [ "uno:GND.1", "neopixels:VSS", "black", [ "v0", "*", "v16" ] ],
    [ "uno:6", "neopixels:DIN", "green", [ "v-16", "*", "v8" ] ],
    [ "uno:5V", "neopixels:VDD", "red", [ "v30", "h-386", "*", "v8" ] ],
    [ "uno:8", "neopixels:DOUT", "green", [ "v-30.77", "h-302.48", "v15.06", "h-67.18" ] ]

```

drf5n, 2023-02-03:
https://wokwi.com/projects/351880975091761743 has a few more attributes:

```
    {
      "type": "wokwi-neopixel-canvas",
      "id": "neopixels",
      "top": -336.88,
      "left": -196.44,
      "attrs": {
        "rows": "16",
        "cols": "16",
        "_temp_gamma": "1",
        "pixelate": "0",
        "matrixLayout": "serpentine"
      }
    }
```

jekader, 2023-10-12:
Thanks to everyone for gathering the info here. Was there any success in getting progressive matrix layout working? I see the `"matrixLayout": "serpentine"` option in the sample above but trying `progressive` or `line` as values does not change anything and serpentine layout is still used.

I've got a physical matrix wired progressively and trying to test firmware for it using wokwi

sutaburosu, 2023-10-12:
> trying `progressive` or `line` as values does not change anything and serpentine layout is still used.

Setting `matrixLayout` to anything except `serpentine` results in a raster scan matrix. You need to restart the sketch for this to take effect.

jekader, 2023-10-12:
thanks for the info, this worked!

BTW, there is a very similar part called `wokwi-neopixel-matrix` which looks more realistic but is less visible.