wokwi/wokwi-features #238: Add a WOKWI_SIMULATOR macro!

arduino12, 2021-12-11:
**Is your feature request related to a problem? Please describe.**
It will be useful for many things..! - for example- when using `tone` -
the first call probably initializing somthing and it takes like 500ms to start producing the audio-
I found out that adding `delay(1000)` to the `setup()` function solve this issue - but it will be much nicer to wrap it in:
```
#ifdef WOKWI_SIMULATOR
delay(1000); // a start-up delay for the simulator to finish initializing audio stuff
#endif
```
So when I flash the same code with the Arduino-IDE - the `delay(1000)` will be omitted out.

**Describe the solution you'd like**
I want the Wokwi web simulator to add (in the background like the `#include <Arduino.h>`) a:
`#define WOKWI_SIMULATOR` or `#define WOKWI_VERSION (2021)` or something like that-
So the code can "know" if it is run inside the simulator or not by checking if this macro is defined..

@urish Many thanks for this geat tool! I have just discovered it today and will sure share my projects with it :)

urish, 2021-12-11:
Thanks for opening the request! This is duplicate of #234, mind adding a comment with the details there?

arduino12, 2021-12-11:
Ohh my bad - I searched for the keyword "simulator" and the #234 issue didn't appeared.
Will add my comment there 👍

closed 2021-12-11 by arduino12