wokwi/wokwi-features #239: Simulação de uma tela de LCD TFT 3.2" com protocolo de comunicação SPI.

jordelsonjunior, 2021-12-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

urish, 2021-12-14:
Thanks for the request Jordel!

Do you have the exact LCD model number that you'd like to see?

We already have the ILI9341 ([docs](https://docs.wokwi.com/parts/wokwi-ili9341) / [docs in Portuguese](https://docs.wokwi.com/pt-BR/parts/wokwi-ili9341))

jordelsonjunior, 2021-12-14:
Eu gostaria de testar a biblioteca TFT_eSPI library, disponível em https://github.com/Bodmer/TFT_eSPI, para esp32 e o LCD mencionado. Seria possível?

urish, 2021-12-16:
We don't have SPI for the ESP32 yet, so I guess it won't work until we add the SPI. 

You can track the status of ESP32 peripherals (including SPI) here: https://docs.wokwi.com/guides/esp32#simulation-features

urish, 2022-03-26:
We now have SPI on the ESP32!

Here's an example with TFT_eSPI: https://wokwi.com/projects/325427785333998162

closed 2022-03-26 by urish

steve-barth, 2022-03-27:
That's great! Thanks!Steve.

FernandoGarcia, 2023-02-07:
Hi @urish!

Thanks for your job!

Could you add "TFT_eWidget" library for this example?

urish, 2023-02-11:
Hello, unfortunately, I do not have the bandwidth to create such example. You can try asking on the [Wokwi discord community chat](https://wokwi.com/discord), people are pretty helpful around there and perhaps someone will be able to take one of the "TFT_eWidget" library examples and run it in Wokwi.