wokwi/wokwi-features #242: Servo motor is not working with ESP32.

omkarshinde22, 2021-12-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Code: 
#include "ESP32_Servo.h"

Servo myservo;  // create servo object to control a servo
                // 16 servo objects can be created on the ESP32

int pos = 0;    // variable to store the servo position
// Recommended PWM GPIO pins on the ESP32 include 2,4,12-19,21-23,25-27,32-33 
int servoPin = 18;

void setup() {
  Serial.begin(115200);
  myservo.attach(servoPin);   // attaches the servo on pin 18 to the servo object
                              // using default min/max of 1000us and 2000us
                              // different servos may require different min/max settings
                              // for an accurate 0 to 180 sweep
  
  Serial.print("Connecting to ");
  
}

void loop() {
  for (pos = 0; pos <= 180; pos += 1) { // goes from 0 degrees to 180 degrees
    // in steps of 1 degree
    myservo.write(pos);              // tell servo to go to position in variable 'pos'
    delay(15);
    Serial.print(pos);                       // waits 15ms for the servo to reach the position
  }
  for (pos = 180; pos >= 0; pos -= 1) { // goes from 180 degrees to 0 degrees
    myservo.write(pos);              // tell servo to go to position in variable 'pos'
    delay(15);                       // waits 15ms for the servo to reach the position
  }
}
[image]

urish, 2021-12-22:
Thanks for the report! This is actually already covered by #214, and is [open for voting](https://wokwi.com/features#feature-1033121859)

closed 2021-12-22 by urish

omkarshinde221, 2021-12-23:
Could we run esp32 and Arduino together? If yes please give one example, where esp32 and Arduino running on same template

urish, 2021-12-23:
> Could we run esp32 and Arduino together?

It should be possible once #186 is done.