wokwi/wokwi-features #243: Add support for NeoPixels with other emitters mix

arduino12, 2021-12-18:
**Is your feature request related to a problem? Please describe.**
NeoPixel elements only support RGB emitters (R=Red, G=Green, B=Blue),
But ther are many other **[LEDs](https://www.aliexpress.com/item/32550663121.html)** and [strips](https://www.aliexpress.com/item/1005001997801363.html) out there with RGBW, RGBWW, RGBNW, WW, CW, NW, WWA emitters!
W=ColdWhite(bluish), WW=WarmWhite(yellowish), NW=NatrualWhite, A=Amber(orange)

[image]

**Describe the solution you'd like**
Add a `type`/`colors`/`emitters` attribute to the NeoPixel LED and strip elements - the default will be RGB,
but the user can set it to RGBW for example -
(uses the SK6812 chipset that "takes" 32bits of color data instead of 24bits of the WS2812 - see the links above)
The letters order is also important - some NeoPixels have different RGB order.
Also adding NeoPixels to the [Docs page](https://docs.wokwi.com/parts/wokwi-rgb-led) will be useful :)

I have the RGBW and WWA LEDs and strip and they are very nice - the white looks much better than the WS2812 that have to combine RGB resulting in 3 separate shadows and poor mixed white color,
The slight disadvantage is that the blue emitter makes the white phosphor to glow a bit..

urish, 2021-12-18:
Thanks for the detailed request, I opened this for [voting](https://wokwi.com/features)

arduino12, 2022-01-19:
Closing this - lets merge it with the previous issue #240 - I renamed it to match the bigger general problem :)

closed 2022-01-19 by arduino12