wokwi/wokwi-features #244: Ability to Edit Wire runs

DiskDoctor, 2021-12-22:


urish, 2021-12-22:
Great, I opened this for [voting](https://wokwi.com/features#feature-1086980412)

leftCoast, 2022-02-08:
Can I add to this a color palette? Maybe a pop up or something? Just a few primary colors would do. Like maybe the resistor colors?

urish, 2022-02-12:
> Can I add to this a color palette? Maybe a pop up or something? Just a few primary colors would do

That makes sense to me. Now let's hope this happens sometimes soon... :)

urish, 2022-04-10:
This issue has been getting a lot of votes recently, so it's likely to make it to the top of the list soon!

urish, 2022-04-18:
Initial version of selecting wires + user interface for setting colors is up!

[image]

The numbers are keyboard shortcuts for the colors (so you can either use the mouse or the keyboard for the first ten colors), and the palette is based on the resistor colors, as suggested. The colors or design of the UI might change, based on the feedback I'll gather.

urish, 2022-04-20:
Beta of wire dragging:

https://wokwi.com/projects/325427785333998162?__beta_dragwires=1

(it works with any project - just add ?__beta_dragwires=1 at the end of a URL)

when you select a wire, you'll see purple dragging handles. Click on them and move the mouse to drag the wire.

Currently missing: grid snapping. 

Also, wires with "*" in the [wire placement instruction list](https://docs.wokwi.com/diagram-format#wire-placement-mini-language) may not drag correctly. But wires created with the UI never have "*" in their placement instruction list, so it should be less of a concern.

djedu28, 2022-04-21:
> Initial version of selecting wires + user interface for setting colors is up!
> 
> [image]

Playing with an old project I saw this new functionality. It turned out great! Congratulations Urish. Thank you so much!!!!

urish, 2022-04-21:
Grid snapping done!

Also improved support for paths with "\*", even though only the segments that come before the \"*\" are currently draggable.

urish, 2022-04-24:
Thanks to @TimMathias, even connections with "*" should drag correctly.

urish, 2022-05-26:
Wire dragging is now generally available.

leftCoast, 2022-05-26:
Can we use the extra cash for this one to add a properties panel that EVERY part uses? I'd LOVE to ditch the entire edit jason thing.

DiskDoctor, 2022-05-27:
> Wire dragging is now generally available.

SWEET! Thank you

urish, 2022-05-28:
>  I'd LOVE to ditch the entire edit jason thing

I think, at least for resistors, LEDs, Push buttons - we should. They are very common.

leftCoast, 2022-05-31:
Well, that's a good start. Is it reasonable to achieve?

urish, 2022-06-06:
I believe so!

urish, 2022-06-16:
LEDs:

[image]

leftCoast, 2022-06-16:
Love it!! Now for buttons?

leftCoast, 2022-06-24:
Looking at this, flip is a good addition! How about adding rotate?

urish, 2022-06-24:
Thanks! Well, once COVID goes away from here :)

leftCoast, 2022-06-25:
You guys still dealing with Covid? Sorry, I didn't know.

MauriceDuteau, 2022-07-03:
I am still learning my way around your system but am very impressed. I have an open source building automation system called T3000, it is fairly active on github. We need many of the features you guys have and are working on. It would be great if we could cooperate. I will learn more about your system and I am sure there will be some areas we can cooperate on.

AriellaE, 2022-07-07:
New interface for editing pushbuttons:

https://user-images.githubusercontent.com/35613240/177746341-5a4cc3b4-1bc5-435f-80b8-4e7c2678a4f4.mp4

leftCoast, 2022-07-07:
That's great! Very handy!

Why are the number of colors to choose for buttons different than for LEDs?

urish, 2022-07-11:
Update: it's now also possible to drag the first/last segments of the wires

urish, 2022-07-11:
Closing, as the requested functionality has been implemented + some nice extras (wire editor, LED editor, push button editor).

closed 2022-07-11 by urish