wokwi/wokwi-features #245: Tutorial Projects area, Organized by topics or tags

DiskDoctor, 2021-12-22:
How about a Tutorial Projects section on the site?

I would be interested in making some basic tutorials for other users, with code snippets that can be copied and used in other projects.

As more projects are added to the site, it will quickly be cumbersome to find examples for learning. 

Tutorials could be tagged with: Tutorial, LED, Switch, WIRELESS, ARRAY, LCD, Neopixel, etc

Could really add to usefulness of site

urish, 2021-12-22:
Thanks for the suggestion! What would such a tutorial might look like? How do you imagine this?

DiskDoctor, 2021-12-22:
A small sketch with comments to indicate which code to copy into your own sketch to use...for instance, debounce would have the debounce code that is used and an example calling it for a button press. Probably have 2 buttons, one with and one without debounce with serial monitor showing status of buttons as they are pressed (HIGH/LOW).

1609 I2C LCD would show what code is used as #define, what is used in setup(), and an example how to use it. 

Sometimes it is challenging to see tutorials or examples and know what code to extract, what libraries are used, etc.

urish, 2021-12-24:
Can you try to make a sample tutorial and share the link here? I'd love to see how this goes, a tutorial section sounds like a great idea to me.

DiskDoctor, 2021-12-24:
Sure. Things are a bit busy now with Christmas and family, but I'll get you something soon.

urish, 2021-12-24:
Amazing. Also, merry Christmas to you and your family!

DiskDoctor, 2021-12-24:
Thank you! Merry Christmas to you and yours as well!

Here is my first go at a Button Tutorial with debounce and interrupt, two of the more troublesome parts to using buttons.

Notes are in the code, including restrictions of which digital pins can be used for interrupts- another Gotcha!

Let me know any suggestions.

Uhm...no way to add text boxes on diagram?

urish, 2021-12-24:
Wow, that was quick!

Mind formatting the code? Use "Alt+Shift+F" in the editor (or press "F1" and select "Format Document")

Text boxes on diagram - see #43 and #222. 

Specifically for LEDs and Push buttons, you can already add a "label" attribute with some text to display below the LED/button. 

Some examples:
- [Bouncing vs non-bouncing buttons](https://wokwi.com/arduino/projects/288681423014986248) - each button has a label below it
- [Arduino Alarm Clock](https://wokwi.com/arduino/projects/297787059514376717) - the Hours / Minutes / Alarm buttons are labeled

DiskDoctor, 2021-12-24:
Thank you. I used format but didn't notice any changes...maybe?

I also added labels for the buttons.

urish, 2021-12-24:
I can notice the changes :-)

Format fixes the indentation / spacing. So instead of 

```cpp
void loop(){
Serial.println("Hello!");
delay(10);
}
```

You get:

```cpp
void loop() {
  Serial.println("Hello!");
  delay(10);
}
```

DiskDoctor, 2021-12-24:
Excellent, thanks. I usually try to have that anyways, must've been lazy today.

I updated the labels (reversed) and the comments in the code to point out the multiple messages and longer LED from the bouncing button that can lead to instability and errors in a program

urish, 2021-12-25:
Somewhat related: https://github.com/wokwi/wokwi-features/issues/100#issuecomment-922047746