wokwi/wokwi-features #246: Add the components in the current zoom window of the simulator instead of default (0,0)

matititam, 2021-12-22:
While adding components, it is comfortable to get the components added in the window of simulator which is in focus (viewable area). If not, I have to always zoom out, go to (0,0) drag the component from there to my current location in the simulator window
[image]

DiskDoctor, 2021-12-23:
This would be an excellent enhancement

urish, 2021-12-24:
Done! Parts are now added to the center (more or less)

closed 2021-12-24 by urish

DiskDoctor, 2021-12-24:
Perfect! It works great! Thank you.

leftCoast, 2021-12-25:
Wait! PLEASE stagger new items. I mean while your on this. Adding them one on top of the other makes people think nothing was added. IE try adding 5 LEDs in a row. Looks like you only got 1.

urish, 2021-12-25:
Yes,  I hear your pain :)

How would you suggest to implement staggering algorithm?

Right now, when you add a new part, we are looking at the screen's center (X/Y), and set the "left" / "top" properties of the new part according to that.

DiskDoctor, 2021-12-25:
> Wait! PLEASE stagger new items. I mean while your on this. Adding them one on top of the other makes people think nothing was added. IE try adding 5 LEDs in a row. Looks like you only got 1.

You better control that trigger finger ;)  But seriously, I've seen this issue with a number of WYSIWYG interfaces, including Visio. I don't think they figured out a simple way to do it, either. 

Unless there is some way to flash an item the moment it is added? That would be the visual cue that a new part was added.

urish, 2021-12-25:
quoting from the discussion on Discord:

> when we add a new part, we currently have the following info:
> - all existing parts and their x/y positions (but not their sizes)
> - x/y positions of viewport (visible area)
> the previous algorithm was:
> `take the minimum of all existing parts x/y`
> so the part would always end up in the top-left corner of the diagram, regarding of your current viewport (visible area)
>
> the new algorithm is:
> `set the top/left to the center of the viewport (visible area)`
>
> how would you suggest to change it?

DiskDoctor, 2021-12-25:
If you have x/y of all parts and visible area of viewport, perhaps you can find the top-est, left-est x/y of any part in the visible area, and any added item will be x+5, y+5 of that part? 

That would stagger any duplicate part added at the same time.

Downside is it will place new parts over top of your current diagram parts. Is that bad? (workaround is to move first part to top left and new parts would start there)

Caveat- if no parts in the visible area, would have to trap that condition and drop the first part in the middle.

What do you think?

urish, 2021-12-25:
Alright, we now have staggering + as @matititam suggested, the parts start at the top left corner (rather than in the middle)

leftCoast, 2021-12-26:
Lets try it and see how it goes.