wokwi/wokwi-features #251: wokwi hungarian translating

steve-barth, 2021-12-26:
Hi! I have included the Hungarian translation for your reference
[wokwi hungarian translating.zip](https://github.com/wokwi/wokwi-features/files/7777543/wokwi.hungarian.translating.zip)

urish, 2021-12-27:
Thank you so much Steve! 

Added the new texts. Can you please take a look here?

https://wokwi.com/arduino/new?lang=hu

steve-barth, 2021-12-27:
Hi Uri!
the following items do not yet appear translated:

userMenu.discord:
- source: Discord
- translation: Vitafórum

userMenu.myProjects:
- source: My projects
- translation: Saját Tervek

userMenu.theClub:
- source: The Club
- translation: Klub

this was not on the translation list:

diagram.addPart.wokwi-dip-switch-8:
- source: DIP Switch 8
- translation: DIP-8 Kapcsolók

( my further suggestion:
sketch -> vázlat,
simulation -> szimuláció,
start the simulation -> szimuláció indítása,
add a new part -> új elem hozzáadása,
zoom, grid and more - nagyítás, rács és egyéb,
docs -> dokumentumok,
put your setup code here, to run once: -> ide írja be a beállítási kódot, ami csak egyszer fut le:,
put your main code here, to run repeatedly: -> ide írja be a fő kódot, ami folyton ismétlődik: )

I’m so glad I can contribute to this great site too.

steve-barth, 2021-12-27:
[mod wokwi hungarian translating.zip](https://github.com/wokwi/wokwi-features/files/7781268/mod.wokwi.hungarian.translating.zip)

urish, 2022-01-04:
Thank you very much Steve, I updated the texts with you latest version. 

I hope to add some of your suggestions in the coming weeks. 

p.s. next time it's enough to send just the .json file, no need for both .json and .txt

closed 2022-01-04 by urish

steve-barth, 2022-01-05:
Thank's Uri!I look forward to the full change!