wokwi/wokwi-features #254: wokwi-tv no longer works with wokwi-pi-pico

sutaburosu, 2022-01-03:
[The tv-pong sketch on Pico](https://wokwi.com/arduino/projects/301402156123030026) used to work. Now it doesn't.

Projects using wokwi-tv on other microcontrollers continue to work fine.

[image]

urish, 2022-01-03:
Thanks for reporting! It doesn't reproduce for me (tried both Chrome and Firefox). Are there any other projects where the issue reproduces for you?

sutaburosu, 2022-01-04:
Adding a TV element to any Pico sketch triggers it for me [even without wiring it up](https://wokwi.com/arduino/projects/319829160639857235).

This only seems to affect Chromium v96. I can't reproduce it in Chrome v96. Very odd.

If I'm reading it correctly it's trying to assign 768 * 576 Uint32s to an array of 767 * 576 * 4 Uint8s, which indeed does not fit. The `width: 767` seems it may be relevant, but I don't know where that is coming from.
[image]

sutaburosu, 2022-01-09:
I'm pretty sure this is caused by a bug in Chromium, so there's no need to have it cluttering up your issue tracker. I can always re-open it if I ever find evidence to implicate Wokwi. :)

closed 2022-01-09 by sutaburosu