wokwi/wokwi-features #256: Request for MLX90614, MAX30100, GPS Neo6m v2

dietherdeyv, 2022-01-04:
**Is your feature request related to a problem? Please describe.**
I wanted to test multiple i2c connections and these components are the most commonly used ones (MLX90614, MAX30100, GPS Neo6m v2)

**Describe the solution you'd like**
It would be a really good starting point to test i2c devices using the simulator and these commonly used i2c devices is a good addition for the web app.

urish, 2022-01-04:
If the goal is just to test multiple i2c connections, there are some popular I2C devices that already exists:

- [DS1307 RTC](https://docs.wokwi.com/parts/wokwi-ds1307)
- [LCD1602 Module](https://docs.wokwi.com/parts/wokwi-lcd1602#i2c-configuration)
- [MPU6050 IMU](https://docs.wokwi.com/parts/wokwi-mpu6050)
- [SSD1306 OLED display](https://docs.wokwi.com/parts/wokwi-ssd1306]

Here's just one fun example from @koepel with a crazy amount of I2C devices: https://wokwi.com/arduino/projects/307886580731740737

urish, 2022-03-26:
Now it's also possible to define custom I2C devices, see #301 for more info. Closing - feel free to open separate issues for specific parts. The GPS is more or less covered by #299.

closed 2022-03-26 by urish