wokwi/wokwi-features #257: issue loading ntp time settings from esp32 .

nielsnl68, 2022-01-05:
**Describe the bug**
I try to get the current date/time using the time lib.  But the esp crashes as soon it set it up.

**To Reproduce**
See the following code:
https://wokwi.com/arduino/projects/319957381134942802

**Expected behavior**
that the esp32 runs without a auto Abort.

**Environment (please complete the following information):**
 - OS: windows10pro
 - chrome
 - Version [e.g. 22]

urish, 2022-01-05:
Reproduces for me:

```
IP address: 10.5.75.167
Ready

Stack smashing protect failure!

abort() was called at PC 0x400dcd44 on core 1

Backtrace:0x4008374e:0x3ffb26800x4008d369:0x3ffb26a0 0x400925b1:0x3ffb26c0 0x400dcd44:0x3ffb2740 0x400d4072:0x3ffb2760 0x400d2535:0x3ffb27d0 0x400d4852:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
```

nielsnl68, 2022-01-05:
Ohhh, yes i forgot that part in my issue.

nielsnl68, 2022-01-05:
I switched the "time.h" lib to the "TimeLib.h" and that seems to work properly.

Although some looks very slow. Not sure of that is the simulator or my code:
https://wokwi.com/arduino/projects/319957381134942802

when you want to close this issue, @urish , then that is fine by me.

urish, 2022-01-05:
> Although some looks very slow. Not sure of that is the simulator or my code:
> https://wokwi.com/arduino/projects/319957381134942802

Thanks for the update! This one doesn't currently compile to me, perhaps you haven't saved the most recent version?

Also, let's keep the issue open, as I want to investigate why this crashed the simulated ESP32. Do you still have the original code that caused the crash?

yepher, 2022-01-06:
Based on the original description above I made this code to reproduce the crash:

https://wokwi.com/arduino/projects/320004099075474004

This is the output:

yepher, 2022-01-06:
tried to load in `gdb` but could not sort out how to make it work:

urish, 2022-01-06:
First of all - thank you!

The [Web version of GDB](https://blog.wokwi.com/running-gdb-in-the-browser/) does not support the ESP32. 

The only way to use GDB with the ESP32 simulator is to install the ESP32 GDB locally and use the [wokwi-gdbserver](https://github.com/wokwi/wokwi-gdbserver) bridge. But then you have to get the ELF file yourself, and it's a pretty complex to set up. 

If you have some experience with building things on Linux and are up to the challenge, I can send you some pointers on how to try to compile the ESP32 GDB to run in the browser. It might be as simple as applying the ESP32 patches over the [existing setup](https://github.com/wokwi/browser-vm-gdb/tree/master), or as complex as fighting strange dependency issues and whatnot.

yepher, 2022-01-06:
I may take you up on getting your notes on how to do that but I think it may be a little bit beyond what I have time to sort out at the moment.  Just trying to help out with low-hanging fruit where I can.

urish, 2022-01-06:
Totally understandable. If you do want to give it a go at some point, #258 has some pointers

urish, 2022-01-18:
Fixed! 

Thanks again @yepher for coming up with a quick reproduction example. Here's the output after fixing the bug:

[image]

closed 2022-01-18 by urish