wokwi/wokwi-features #260: German Translation

joe-hidden, 2022-01-07:
{
  "userMenu.signIn": "Anmelden",
  "userMenu.signUp": "Registrieren",
  "userMenu.discord": "Discord",
  "userMenu.myProjects": "Meine Projekte",
  "userMenu.theClub": "Der Club",
  "userMenu.featureRoadmap": "Feature Roadmap",
  "userMenu.logout": "Abmelden",
  "project.header.save": "Speichern",
  "project.header.share": "Teilen",
  "project.header.likeProjectTooltip": "Like zu diesem Projekt hinzufügen",
  "project.header.unlikeProjectTooltip": "Like von diesem Projekt entfernen",
  "project.menu.saveACopy": "Kopie speichern",
  "project.menu.lockProject": "Projekt sperren",
  "project.menu.unlistProject": "Projekt entfernen",
  "project.menu.downloadProjectZip": "Projekt ZIP herunterladen",
  "project.error.saveFailed": "Speichern fehlgeschlagen: Das Projekt ist gesperrt!",
  "project.error.signInToLikeProject": "Bitte anmelden um dem Projekt ein like hinzuzufügen",
  "simulator.buildFailedDialog.title": "Build fehlgeschlagen!",
  "simulator.downloadingDataDialog.title": "SD Kartendaten werden geladen...",
  "editor.fileMenu.rename": "Umbenennen",
  "editor.fileMenu.delete": "Löschen",
  "editor.fileMenu.newFile": "Neue Datei...",
  "editor.fileMenu.uploadFile": "Datei(en) hochladen...",
  "editor.fileMenu.confirmDelete": "Die Datei {{filename}} wird gelöscht. Wirklich fortfahren?",1111
  "editor.fileMenu.error.invalidFileName": "Ungültiger Dateiname: Zeichen {{char}} ist nicht zulässig.",
  "editor.fileMenu.error.invalidFileType": "Ungültiger Dateityp. Unterstützte Dateitypen: {{types}}.",
  "editor.fileMenu.error.fileExists": "Eine Datei mit diesem Namen existiert bereits",
  "editor.fileMenu.error.uploadFailed": "Einige Dateien konnten nicht hochgeladen werden!",
  "editor.sdCard.uploadFiles": "Dateien hochladen",
  "editor.sdCard.uploadCompleteFolder": "Ganzen Ordner hochladen",
  "editor.sdCard.deleteAllFiles": "Alle Dateien löschen",
  "editor.sdCard.noFilesYet": "Noch keinen Dateien vorhanden. Bitte erst welche hochladen.",
  "diagram.menu.fit": "Anpassen",
  "diagram.menu.zoomIn": "Vergrößern",
  "diagram.menu.zoomOut": "Verkleinern",
  "diagram.menu.toggleGrid": "Gitter umschalten",
  "diagram.menu.fullScreen": "Vollbild",
  "diagram.menu.exitFullScreen": "Vollbild beenden",
  "diagram.menu.help": "Hilfe",
  "diagram.addPart.wokwi-arduino-uno": "Arduino Uno",
  "diagram.addPart.wokwi-arduino-mega": "Arduino Mega 2560",
  "diagram.addPart.wokwi-arduino-nano": "Arduino Nano",
  "diagram.addPart.wokwi-pi-pico": "Raspberry Pi Pico",
  "diagram.addPart.wokwi-attiny85": "ATtiny85",
  "diagram.addPart.wokwi-led": "LED",
  "diagram.addPart.wokwi-pushbutton": "Taster",
  "diagram.addPart.wokwi-resistor": "Widerstand",
  "diagram.addPart.wokwi-slide-switch": "Schiebeschalter",
  "diagram.addPart.wokwi-membrane-keypad": "Tastatur",
  "diagram.addPart.wokwi-logic-analyzer": "Logikanalysator (8 Kanäle)",
  "diagram.addPart.wokwi-max7219-matrix": "LED Punktmatrix (MAX7219)",
  "diagram.addPart.wokwi-led-ring": "NeoPixel Ring",
  "diagram.addPart.wokwi-neopixel-meter": "NeoPixel Streifen",
  "diagram.addPart.wokwi-buzzer": "Piepser",
  "diagram.addPart.wokwi-dht22": "DHT22",
  "diagram.addPart.wokwi-potentiometer": "Potentiometer",
  "diagram.addPart.wokwi-slide-potentiometer": "Schiebepotentiometer",
  "diagram.addPart.wokwi-lcd1602": "LCD 16x2",
  "diagram.addPart.wokwi-lcd1602-i2c": "LCD 16x2 (I2C)",
  "diagram.addPart.wokwi-lcd2004": "LCD 20x4",
  "diagram.addPart.wokwi-lcd2004-i2c": "LCD 20x4 (I2C)",
  "diagram.addPart.wokwi-servo": "Servo",
  "diagram.addPart.wokwi-rotary-dialer": "Drehknopf",
  "diagram.addPart.wokwi-ssd1306": "SSD1306 OLED Anzeige",
  "diagram.addPart.wokwi-ds1307": "DS1307 RTC",
  "diagram.addPart.wokwi-hc-sr04": "HC-SR04 Ultraschall Sensor",
  "diagram.addPart.wokwi-rgb-led": "RGB LED",
  "diagram.addPart.wokwi-7segment": "7 Segment Anzeige",
  "diagram.addPart.wokwi-7segment-4digits": "7 Segment Anzeige (4 -stellig)",
  "diagram.addPart.wokwi-led-bar-graph": "LED Balkenanzeige",
  "diagram.addPart.wokwi-analog-joystick": "Analoger Joystick",
  "diagram.addPart.wokwi-ir-receiver": "IR Empfänger",
  "diagram.addPart.wokwi-ir-remote": "IR Fernbedienung",
  "diagram.addPart.wokwi-pir-motion-sensor": "PIR Bewegungssensor",
  "diagram.addPart.wokwi-ntc-temperature-sensor": "Analoger Temperatursensor (NTC)",
  "diagram.addPart.wokwi-74hc595": "74HC595 Schieberegister",
  "diagram.addPart.wokwi-franzininho": "Franzininho",
  "diagram.addPart.wokwi-photoresistor-sensor": "Fotowiderstand (LDR)",
  "diagram.addPart.wokwi-ky-040": "KY-040 Drehencoder",
  "diagram.addPart.wokwi-mpu6050": "MPU6050 Beschleunigungssensor + Gyroskop",
  "diagram.addPart.wokwi-microsd-card": "microSD Karte"
}

urish, 2022-01-08:
Thank you so much Joe! 

I uploaded the new texts, mind having a look?

https://wokwi.com/arduino/new?lang=de

joe-hidden, 2022-01-09:
Was my pleasure. Looksgreat, only the Library Manager had no texts to
translate (yet?)

urish, 2022-01-09:
Thanks!

Yes, it's a new feature, so the text might change over the next few days. Once it stablilizes I'll ping over #221

closed 2022-01-09 by urish