wokwi/wokwi-features #261: Code window loses horizontal scroll below about 50%

DiskDoctor, 2022-01-08:
When split screen between code window and diagram window is changed by dragging vertical split to the left, as code window drops below a certain amount (50%?) or more, the horizontal scroll bar in the code window disappears.

Problem: Cannot scroll and see code in the window

urish, 2022-01-08:
Doesn't reproduce for me. It could be that I do not understand the instructions very well.

What browser do you use?
Mind sharing a screen recording showing the issue in action?

DiskDoctor, 2022-01-08:
Two screenshots. First has the horizontal scroll bar at bottom of code window.

Second shows scroll bar missing at the bottom of the code window. Appears when the vertical split is dragged left until the TABS at the top of the code window are wrapped, then the scroll bar disappears, too.
[image]

Win 10 pro, Chrome
[image]
.

sutaburosu, 2022-01-08:
This happens in both Chromium and Firefox for me. The key to triggering it is to make the list of tabs span onto another line. I guess the scrollbar is scrolling off the bottom of the view port. Temporarily resizing the browser window brings the scrollbar back.

DiskDoctor, 2022-01-08:
I agree. Resizing larger OR smaller brings back scrollbar. Can it be fixed?

urish, 2022-01-08:
> The key to triggering it is to make the list of tabs span onto another line

Good catch. Now this reproduces for me!

urish, 2022-01-08:
Uploaded a fix - can you please test and report back?

sutaburosu, 2022-01-08:
Works great here for me. Thanks, Uri. :heart:

urish, 2022-01-08:
Great, thanks for helping me figure this out!

closed 2022-01-08 by urish

DiskDoctor, 2022-01-08:
Looks good here! Thanks