wokwi/wokwi-features #262: Orange rendering of Neopixels is yellow?

DiskDoctor, 2022-01-08:
0xff,0x7f,0x00 should be orange, but it shows up in simulator as yellow. 

Used this reference for colors: [https://i.pinimg.com/originals/d8/81/b2/d881b2850db572a124dc5c17549c40d6.png](url)

urish, 2022-01-08:
Can you share the exact code you are running in the simulator?

Some libraries (e.g. FastLED) has built-in color correction, which may affect the color rendering

DiskDoctor, 2022-01-08:
This Christmas Star project: [https://wokwi.com/arduino/projects/318548233369420372](url)

Line 183 is an example, but anywhere with 'orange' rgb code (colors are in comments)

sutaburosu, 2022-01-08:
This is because you have `"matrixBrightness": "10"` on your LEDs. That multiplies each of the RGB values by 10 (and then clamps them to a valid range), so an input of 0x7f becomes 0xff.

urish, 2022-01-08:
Cool project :)

DiskDoctor, 2022-01-08:
> This is because you have `"matrixBrightness": "10"` on your LEDs. That multiplies each of the RGB values by 10 (and then clamps them to a valid range), so an input of 0x7f becomes 0xff.

So what is the correction? "matrixBrightness": "0" ?? I tried it and do get more orange.

But I also set another leg at 90, no real change. Or is it hex? I don't have any documentation on that element.

Is it a rendering onscreen color issue, or what is best way to correct?

sutaburosu, 2022-01-08:
A value of "1" would present the colours FastLED is sending unchanged.

DiskDoctor, 2022-01-08:
> Cool project :)

Thanks. Still working on input, but can manually change from switch for single effect, step through in order, or randomize effects. Adapted some effects from allLEDeffects (or something like that) and tweaked for the effect we wanted.

If you want to test, you can set it to start at an effect, which is in the array or the giant if..then..else section

DiskDoctor, 2022-01-08:
> A value of "1" would present the colours FastLED is sending unchanged.

YES! That did make orange orange. White is dull/grayish. Not a big deal in the simulator, the LEDs are slightly different anyways.

Can you tell me the options for this setting and what they mean? Or the range of settings?

sutaburosu, 2022-01-08:
Yes, it is just a multiplication factor. 

If you were using the "matrixBrightness" to try to match the appearance of real-world LEDs more closely in the simulator, you might want to try the beta functionality of gamma correction instead. Assuming your monitor has a gamma exponent of 2.2,  you would use a value of 1/2.2 = 0.45.  So instead of "matrixBrightness" try `"_temp_gamma": "0.45"`.

Also, to disable FastLED's colour correction, modify your addLeds call in setup() to `setCorrection(UncorrectedColor);`. That usually gives better results on monitors.

DiskDoctor, 2022-01-08:
> Yes, it is just a multiplication factor.
>
> If you were using the "matrixBrightness" to try to match the appearance of real-world LEDs more closely in the simulator, you might want to try the beta [...]

I'll give that a shot. Thanks.

I guess that means the "fix" for this issue is to add documentation to the attrs when the neopixel is official on the DOCS page ?

DiskDoctor, 2022-01-08:
Using the gamma setting and the UncorrectedColor do look significantly better. Thank you!

sutaburosu, 2022-01-08:
No problem. Two other tips that might help you: add `?dark=1` to the end of the project URL to get a darker background, and to hide the wires set their colour to be an empty string `""`. :)

DiskDoctor, 2022-01-08:
> ?dark=1

Thanks. This blinks momentarily with a solid dark black, canvas, but as soon as the elements are displayed, it is back to grey. Something you want me to test as this is an experimental feature, right?

Hiding the wires is a good trick. I thought leaving them visible can help others understand what I did. 

Actually- I used the diagram.json tab to make the connections and let it route where it wanted. I was impressed with the wire routing. 

Also had to flip a bunch of the neopixels, the wire connections and head/tail all look the same right now. Would be good to distinguish them eventually ?

DiskDoctor, 2022-01-08:
I did hide all the interconnecting wires on the neopixels. Kept the ones from the mega. You're right, it does look much better now. Thanks!

sutaburosu, 2022-01-08:
> back to grey

It sounds as though you're already running Wokwi's dark theme then. I was seeing [quite a bright background](https://wokwi.com/arduino/projects/318548233369420372?dark=0), which kinda spoiled the LED effects.

> Also had to flip a bunch of the neopixels, the wire connections and head/tail all look the same right now. Would be good to distinguish them eventually ?

I guess whilst the sketch isn't running, some visual indication of where the first pixel is could be useful. Is that what you meant?

DiskDoctor, 2022-01-08:
Yes, which side is first pixel would be helpful. And some better contrast or indication where the connectors are. Maybe a lighter section in the middle, surrounding the connections, to make them standout?

Especially when zoomed out for something like this, it was very hard to find and make the starting connection (zoom in), then try to find the opposing connection while zooming out. 

If you are dragging a wire and need to drag/pan the view, it drops a wire turn point. Another reason  I did it through diagram.json.

sutaburosu, 2022-01-08:
> If you are dragging a wire and need to drag/pan the view, it drops a wire turn point

Ah! I've just realised that I unconsciously work around that particular GUI foible. To pan whilst routing a wire I zoom out and then then zoom in again on a different area. I feel we may be getting very off-topic for this issue tracker. Why don't you join us on the [Wokwi Discord server](https://wokwi.com/discord), where this kind of chat is welcomed?

DiskDoctor, 2022-01-08:
Done. Thanks ;)

sutaburosu, 2022-01-21:
@DiskDoctor does this bug need to remain open? I just skimmed through to remind myself of what was discussed, and I can't see any bug in Wokwi here. Would you please close it if you agree. Thanks.

DiskDoctor, 2022-01-22:
Documentation of this tweak would fix the issue, but we can close this bug report- just leave it here for anyone to read about the tweak until Wokwi catches up with documentation. I agree with the current approach of Function first, docs later.

closed 2022-01-22 by DiskDoctor