wokwi/wokwi-features #263: Can you add a random generator for use as randomSeed() ?

DiskDoctor, 2022-01-08:
A common trick to get a randomSeed for a random number is to use analogRead(A4) (or other unused analog pin) that fluctuates do to interference or other factors. 

The analog pins don't seem to have this effect in the simulator. I tested with A4.

Perhaps an attrs in the Boards themselves for a "Random pin" that itself is a random number from say Seconds of the current network time out to 4 decimal places.

Even if it only updated the random number each time Run is hit, it would be a help. Won't re-randomize during execution of a program, but not sure if that is possible. After all it is a simulator, won't be exact.

urish, 2022-01-08:
Thanks for reporting, as far as I can see analogRead() from a disconnected pin does return different numbers every time:

https://wokwi.com/arduino/projects/320194639812362834

Do you see something different?

DiskDoctor, 2022-01-08:
Thank you for the example. In my haste, I neglected to set A0 as input.

This now works: 

pinMode(A4, INPUT);
randomSeed(analogRead(A4));

Thank you. This thread can be closed.

urish, 2022-01-08:
Okay, thanks!

closed 2022-01-08 by urish