wokwi/wokwi-features #264: Zoom out with mouse scroll doesn't center on pointer

DiskDoctor, 2022-01-08:
When zooming in, the view properly zooms on the pointer location.

But zooming out, the view does not zoom out on the cursor position.

Using the mouse scroll button.

sutaburosu, 2022-01-08:
This seems to behave as I would expect. Are you seeing different results to me?

https://user-images.githubusercontent.com/20606/148635444-0d4e8169-84ef-4628-8dfe-0c34342e2517.mp4

sutaburosu, 2022-01-08:
From Discord:
> **DiskDoctor:** Hmm...maybe that is right. I was trying Visio. When getting to edge of screen, it pans the screen, so perhaps that was happening at the same time. Not sure.
> **sutaburosu:** I just tested a couple of image editors and LibreOffice Draw. They all work the same way as Wokwi.
> **DiskDoctor:** Okay. I must be looking at it wrong. I am seeing the same thing. Maybe it should be a request to pan when dragging against the edges instead. That seems to be the helpful function.

urish, 2022-01-10:
So should we keep this open? or close?

sutaburosu, 2022-01-10:
I vote to close this bug. The feature request for pan when reaching the edge would be a new feature request ticket, if @DiskDoctor still wants it?

DiskDoctor, 2022-01-11:
Yes, panning when at the edge of the viewport is more accurate. Closing.

DiskDoctor, 2022-01-11:
Closed in favor of new request for panning the screen instead

closed 2022-01-11 by DiskDoctor