wokwi/wokwi-features #266: DHT is not working with ESP312

yepher, 2022-01-08:
**Describe the bug**
DHT22 is producing data but ESP32 is not decoding it

**To Reproduce**

* [Sample Project](https://wokwi.com/arduino/projects/320219663593112146)

1. Create ESP32 project on Wokwi
2. Add a DHT22 and wire:
    * VCC->DHT VCC (pin1)
    * VCC->10KΩ resiter pullup to DHT Data (pin2)
    * DHT Data (pin2) -> D4
    * GND->DHT22 GND (pin3)
3. Copy sample code from [Adafruit DHT-sensor-library]( https://github.com/adafruit/DHT-sensor-library/tree/master/examples)
4. Run project and note output reports `Error reading...`

**Expected behavior**

Read data from the sensor and report to the console.

**Environment (please complete the following information):**
 - OS: MacOS
 - Browser [chrome and safari]

**Additional context**

Looks like DHT22 is doing the right thing and that the ESP32 is not interpreting it properly

[image]

yepher, 2022-01-08:
This comment in the DHT lib may provide a clue

Summary: use `digitalRead(_pin)` instead of `(*portInputRegister(_port) & _bit)`

```
// Otherwise fall back to using digitalRead (this seems to be necessary on
// ESP8266 right now, perhaps bugs in direct port access functions?).
```

[DHT.cpp#L378](https://github.com/adafruit/DHT-sensor-library/blob/master/DHT.cpp#L378]

yepher, 2022-01-08:
I created this code to try and isolate the issue:

[DHT22 Hacked Demo](https://wokwi.com/arduino/projects/320228847292252754)

Mine timesout on line 91 when `i=3`

yepher, 2022-01-08:
Added pulse in expect.

[Wokwi Project](https://wokwi.com/arduino/projects/320228847292252754)

Expect Logic:

```
uint32_t expectPulse(bool level) {

  uint32_t count = 0;

  digitalWrite(ESP_OUT_REF, HIGH);
  digitalWrite(ESP_OUT_REF, LOW);
  
  // Otherwise fall back to using digitalRead (this seems to be necessary on
  // ESP8266 right now, perhaps bugs in direct port access functions?).
  while (digitalRead(DHTPIN) == level) {
    if (count++ >= _maxcycles) {
      return TIMEOUT; // Exceeded timeout, fail.
    }
  }

  digitalWrite(ESP_OUT_REF, HIGH);
  digitalWrite(ESP_OUT_REF, LOW);

  return count;
}

```

```
Max Timeout: 4294967295
pullTime: 55
maxCycles: 80000
Failed: cycles[78]

Cycles Dump

0x10 0x7 0x10 0x7 0x10 0x7 0x10 0x7 0x10 0x6 0x11 0x6 0x12 0x8 0x36 
0xA 0x9 0x19 0x10 0x6 0x11 0x6 0x10 0x19 0x10 0x7 0x10 0x7 0x10 0x7 
0x10 0x7 0x10 0x6 0x11 0x8 0x29 0x14 0x1C 0x6 0x10 0x7 0x10 0x6 0x11 
0x6 0x10 0x7 0x10 0x19 0x10 0x7 0x10 0x7 0x10 0x19 0x10 0x38 0x19 0x4 
0x10 0x7 0x10 0x19 0x10 0x19 0x10 0x6 0x11 0x18 0x10 0x19 0x10 0x19 0x10 
0x2F 0x23 0x3
```

[image]

[expect_ref_signal.vcd.zip](https://github.com/wokwi/wokwi-features/files/7833535/expect_ref_signal.vcd.zip)

urish, 2022-01-08:
Thanks for digging into this! Seems like the code is misses some of the pulses from the DHT sensor - there are several spot where you can see some data on the DHT line, but the ESP Ref spot misses them.

lazaroofarrill, 2022-01-16:
I've had lots of trouble using DHT library with an actual ESP32.
[This library](https://github.com/beegee-tokyo/DHTesp) solved it.

urish, 2022-01-22:
Testing the DHT Unified Sensor library on an actual ESP32 with a DHT22 sensor:

[image]

Test setup:

[image]

This is still better than the simulation - in the simulation it always fails reading from the sensor, in real life about 10-15% of the time.

urish, 2022-02-01:
Closing this is this seems to be a problem of the library, and not the simulator.

For anyone who is trying to use the DHT22 on the ESP32, here's the "DHT sensor library for ESPx" suggested by @lazaroofarrill that does work. Here's an example project (I'll also put the link in the docs):

https://wokwi.com/arduino/projects/322410731508073042

closed 2022-02-01 by urish