wokwi/wokwi-features #267: SSL Support for the ESP32

urish, 2022-01-09:
SSL support is required for HTTPS. Most Web APIs today use HTTPS (e.g. the Telegram API), so having support for it in Wokwi will be very useful.

urish, 2022-01-22:
Some progress update: implemented AES and SHA Accelerators for the ESP32. Here are some demos (I used these to verify the behavior against the physical hardware):

- [AES accelerator in ESP32 Arduino](https://wokwi.com/arduino/projects/321485575446069843)
- [AES accelerator in ESP32 MicroPython](https://wokwi.com/arduino/projects/321484545174012499)
- [SHA accelerator in ESP32 MicroPython](https://wokwi.com/arduino/projects/321485907801670226)

We're still missing the RSA Accelerator peripheral, which is also required for SSL/TLS to work in the simulation.

urish, 2022-01-26:
Update: There's an initial version of the RSA accelerator.

For the curious, here's the MicroPython script that I used for verifying its operation (comparing the outputs VS the real ESP32 device):

https://wokwi.com/arduino/projects/321803425724498516

Now that we have the AES, SHA and RSA accelerators, we're getting really close to TLS (HTTPS) support. Here's an example using the Arduino core. It occasionally works, but requires a lot of patience. It can probably can faster after I optimize the AES/RSA accelerator implementation:

* [Example of HTTPS Request from the ESP32 Arduino Core](https://wokwi.com/arduino/projects/321852683139940948)
* [MicroPython ESP32 HTTPS example](https://wokwi.com/arduino/projects/321853585899913810)

The MicroPython example currently fails with:

```
  File "main.py", line 13, in <module>
  File "urequests.py", line 116, in get
  File "urequests.py", line 62, in request
OSError: (-30592, 'MBEDTLS_ERR_SSL_FATAL_ALERT_MESSAGE')
```

Looking at the Wireshark (PCAP) dump, we can see the TLS connection is established, but immediately aborted due to decrypt error:

[image]

urish, 2022-01-31:
Note: spotted the issue that caused the TLS failures: the SHA hardware accelerator shares the state between SHA384 and SHA512, and this behavior wasn't documented. I fixed the implementation, and now TLS works in both Arduino and MicroPython!

Sample projects:

* [HTTPS in the ESP32 Arduino Core](https://wokwi.com/arduino/projects/321852683139940948)
* [MicroPython ESP32 HTTPS example](https://wokwi.com/arduino/projects/321853585899913810)

Many thanks to everyone who voted and supported this feature.

closed 2022-01-31 by urish

danicampora, 2022-11-02:
@urish I'm interested in how you fixed this issue, could you please share the code patch? Thanks.

urish, 2022-11-03:
Hi @danicampora, that was something internal to the simulator. The SHA accelerator has 16 bytes that stores the internal state (that is part of the SHA512 algorithm).  The ESP32 supports several variants of SHA: SHA1, SHA256, SHA384, and SHA512. Initially, I assumed that each variant has a different state storage, but it turned out that SHA384 and SHA512 share the initial state. Thus, if you started a SHA384 operation, but then asked the SHA accelerator to continue with SHA512, you'd have different results on the hardware compared to the simulator. The fix was simply to use the same 16 byte buffer to store the state for both SHA384 and SHA512.

You can see the MicroPython script that I used to verify this behavior here: https://wokwi.com/projects/321971676922249810

danicampora, 2022-11-03:
Hi @urish, understood, thank you!

urish, 2022-11-03:
You're welcome! I'm curious, what project are you working on?