wokwi/wokwi-features #270: Integrate Kalume + support JS files

drohen, 2022-01-11:
Wokwi doesn't have a JS runtime for RP2040, and also the code editor doesn't support JS files. Implement loading Kaluma-based projects along with allowing JS files and maybe auto-complete for the Kaluma/JerryScript API.

urish, 2022-01-11:
some implementation notes:

[Kaluma Flash Layout](https://github.com/kaluma-project/kaluma/blob/af505a888ac93c4f11e21b5f15c1b1790000309d/targets/rp2/boards/pico/board.h#L37-L45). The code is stored separately from the rest of the file system (there's 512kb for code storage, 512kb for the filesystem). It is stored as a simple null-terminated string.

urish, 2022-01-13:
Well, I managed to hack something together so hopefully it'll be useful enough for your talk:

https://wokwi.com/arduino/new?template=kaluma-pi-pico

drohen, 2022-01-13:
Ok, so it works - which is so great! I'll keep appending to this issue any bugs I find.

drohen, 2022-01-13:
For some reason it does a soft reboot immediately after loading, so I can get a single value out, but then I need to enter: `.load code.js` in the REPL to get it running again.

urish, 2022-01-13:
What if you add a delay(1000); at the end of the code?

drohen, 2022-01-13:
delay doesn't help

urish, 2022-01-13:
There you go: https://wokwi.com/arduino/projects/320703884609716818

urish, 2022-01-13:
That's a good question!

drohen, 2022-01-13:
I can work with what there is for now anyway, I don't see anything wrong with having a loop with the delay call. If you find it's something to do with Kaluma, I'm sure the dev would be interested. Cheers

urish, 2022-01-13:
Seems like it was Wokwi. Try your original example again?

drohen, 2022-01-13:
Yep works as expected!

urish, 2025-01-13:
Kaluma support was removed from Wokwi due to low usage.

closed 2025-01-13 by urish