wokwi/wokwi-features #271: Missing German translations

lazyzero, 2022-01-13:
project.shareDialog.title:
- source: Share Project
- translation: 

project.shareDialog.copyToClipboard:
- source: Copy to clipboard
- translation: In die Zwischenablage kopieren

project.shareDialog.pleaseSave:
- source: To share this project, please save it first:
- translation: Um diese Projekt zu teilen, bitte zuerst speichern:

project.shareDialog.saveButton:
- source: Save projet
- translation: Projekt speichern

project.shareDialog.closeButton:
- source: Close
- translation: Schliessen

AriellaE, 2022-01-13:
Thank you @lazyzero!
No need to translate project.shareDialog.title: (source: Share Project)?

urish, 2022-01-15:
Translations updated. Thanks!

closed 2022-01-15 by urish