wokwi/wokwi-features #274: Sming Framework based applications

slaff, 2022-01-13:
[Sming Framework](https://github.com/SmingHub/Sming) is a open source embedded C/C++ framework with Arduino compatibility.
It supports multiple architectures such as Esp32, Esp8266 and RP2040 among others. And it would be awesome if the Sming users and developers can design and test their work on Wokwi simulator before flashing it on their micro-controllers.

As far as I can see at the moment there are "templates", ex: https://wokwi.com/arduino/new?template=micropython-esp32, that describe the preinstalled software and emulator. Can you give us hints how to add Sming as a template?

urish, 2022-01-13:
Can one use the Arduino toolchain to build Sming projects? 

Or does it need a different setup?

urish, 2022-01-13:
Looking at the documentation (and your comments), it looks like Sming does need its own build environment - you can't build Sming projects only using the Arduino CLI. Is that correct?

slaff, 2022-01-13:
>  Is that correct?

Yep, this is correct.

urish, 2022-01-13:
Got it. The main issue is that the build server only supports the Arduino CLI at the moment. There's been some [interesting discussion on discord](https://discord.com/channels/787627282663211009/787630013658824707/926591631149383700) about adding support to third party build processes, with the two main points being security and performance.

What makes you interested in getting Wokwi to support Sming projects?

slaff, 2022-01-13:
> What makes you interested in getting Wokwi to support Sming projects?

It looks like a great fit for designing and testing applications on Esp32's emulator. Should save development time,  spare the Flash memory on the real device and in addition make it easy to visually demonstrate and share a device or/and application. For development we have already a [Host emulator](https://sming.readthedocs.io/en/latest/arch/host/index.html) that can help the developer create and test their code. But it cannot help them test integration with sensors. And here is where your emulator shines :)

urish, 2022-01-16:
> Interestingly enough the simulation is working :)

Happy to hear that!

> Will the emulator work if I pack them in one file with the appropriate offsets? What I mean is something like:

It might - the provided bin file is loaded at flash address 0x10000, unless the simulator detects there's a boot loader at the beginning, and then it's loaded at 0x0 or 0x10000. Worth giving it a shot.

> Slightly off-topic: It would be great if we can upload two mandatory things and one optional.

There's `diagram.json` file which specifies is the sensors, parts and connection. Is that what you mean?

As for ram.bin, how would that be generated? Can you tell me more about the use cases?

urish, 2022-02-26:
@slaff have you lost interest?

slaff, 2022-02-28:
> have you lost interest?

No, not at all. At the moment Sming binary firmware compiled on the PC could run on the web emulator. But it feels like second hand citizen. 

Can we help you build your docker image or whatever containers you are using so that your users can compile Sming applications directly in the browser? For a minimum compilation Sming will need a compiler(already available with ESP32 arduino), `make`, git and python 3 for some examples. The latter might also be optional. What differentiates Sming from Arduino for example in terms of compilation is that every library and Sming itself can be compiled with multiple different options and directives and that makes providing pre-compiled libraries for all library variants almost impossible.  
@mikee47 any ideas on providing a `minimum` Sming docker image?

mikee47, 2022-02-28:
> ... @mikee47 any ideas on providing a `minimum` Sming docker image?

@slaff https://github.com/SmingHub/Sming/tree/develop/Tools/Docker/cli has the necessary

urish, 2022-02-28:
Thanks! I'm looking into this now. The main challenges would probably be ensuring that:
1. The builds go fast. If a build takes more than a couple of seconds, it can be very frustrating for the users.
2. The builds environment is secure. We definitely do not want bad actors to run whatever stuff they want on our servers.

urish, 2023-11-13:
Closing, as users can now use [Wokwi for VS Code](https://docs.wokwi.com/vscode/getting-started) to run their sming applications.

closed 2023-11-13 by urish

slaff, 2023-12-18:
@urish we have integrated Wowki in Sming Framework. The documentation is here: https://sming.readthedocs.io/en/latest/experimental/wokwi.html.  It would be great if you can link this information to Wokwi's documentation: https://docs.wokwi.com/guides/esp32.

urish, 2023-12-18:
Lovely, thank you! I added a link. Also, tried it - works great!

[image]

urish, 2023-12-18:
One thing I noticed: in some cases, I had to run `make ide-vscode ENABLE_WOKWI=1 SMING_ARCH=Esp32` twice in order for the wokwi files to be generated

slaff, 2023-12-19:
> in some cases, I had to run make ide-vscode ENABLE_WOKWI=1 SMING_ARCH=Esp32 twice

@urish thank you for reporting. I will investigate it...