wokwi/wokwi-features #275: Adding Blues Wireless Notecard SoM

rdlauer, 2022-01-13:
We would love to see the Blues Wireless Notecard added to Wokwi for use in prototyping cellular connections and relaying sensor data to the cloud.

urish, 2022-01-14:
Looking at the difference available modules, seems like adding the [Notecarrier-B](https://dev.blues.io/hardware/notecarrier-datasheet/notecarrier-b/) (with an on-board Notecard) would be a good option - as it'll allow users to experiment with an MCU of their choice (e.g. Arduino Uno, Arduino Mega, Raspberry Pi Pico, ESP32 and others in the future). Or do you have other thoughts about this?

Do you happen to have an SVG drawing of the Notecarrier-B with the Notecard mounted on?

yepher, 2022-01-14:
If it becomes a bottleneck and you can't find an SVG I can draw it for you.

urish, 2022-01-14:
Thanks @yepher! I'll let you know

rdlauer, 2022-01-14:
@urish We have a repo that contains illustrations of the Notecard and Notecarriers here: https://github.com/blues/note-hardware

Notecarrier-B is a fine place to start, I would say the AL might be better in terms of it being more commonly used, but for the purposes of Wokwi it shouldn't matter much.

urish, 2022-01-14:
We can do the AL then, I guess there's no big difference in terms of how you'd use it. 

Do you have any idea which interface is more common, the Serial (UART) or I2C?

rdlauer, 2022-01-14:
Probably I2C

urish, 2022-01-14:
Great! Brandon mentioned you may be able to recreate the JSON API parser + glue code that connects it to the blues cloud in a different language. JavaScript (specifically TypeScript) will be best, if that's an option.

rdlauer, 2022-01-14:
Yeah absolutely...this will take some discussion on our end, but we will plan on TypeScript and I'll update you here.

urish, 2022-01-14:
Cool, let me know if you need any inputs from me

urish, 2022-03-18:
Initial implementation: https://wokwi.com/projects/321614504765948499

closed 2022-03-18 by urish

rdlauer, 2022-04-21:
@urish here is an updated svg for our new Notecarrier-A with a Notecard on it
[image]

urish, 2022-04-21:
Thanks, looks so much better than what we have at the moment! 

Sent you some feedback + questions on Discord.