wokwi/wokwi-features #276: upip support for micropython

traverseda, 2022-01-15:
One of the main selling points of micropython is the ability to dynamically load code. Support for upip, from the library manager and/or from inside micropython itself would be very useful.

urish, 2022-01-15:
Would adding a requirements.txt that will installed the packages when your project is running do the trick?

traverseda, 2022-01-15:
I think it's reasonable to separate it into two separate issues, one for static `requirements.txt` based modules and one for dynamically loaded modules. My understanding is that currently the flash isn't writable? I think being able to statically add libraries to requirements.txt is more important.

Eventually I'd like to include an "app store" type feature, which would need to be able to install upip packages at run time, but it's not a high priority.

jonnor, 2026-05-25:
Using `mip.install()` to install [native C modules](https://docs.micropython.org/en/latest/develop/natmod.html) in Wokwi also works! Example for ESP32 at: https://wokwi.com/projects/465000960050404353

I tested this using one of the modules from https://github.com/emlearn/emlearn-micropython - a Machine Learning and Digital Signal Processing for MicroPython.