wokwi/wokwi-features #277: Time.h on ESP32 gives hours > 24

sutaburosu, 2022-01-17:
Benjamin@Discord reported that at 2021-01-17T16:44:00 [this sketch](https://wokwi.com/arduino/projects/321044828657812052) shows:
```
Sunday, July 28 1986 310985:36:29
Hour: 310985
```

Adding 310,985 hours to 1986-07-26 gives the correct time at the moment of running that sketch.

This was double-checked in Wokwi using the 'Time' library along with `#include <TimeLib.h>`. @Yepher kindly tested this on real hardware, and confirmed that the results in Wokwi differ. When they ran that sketch on real hardware their output was:
```
11:54:31.640 -> Monday, January 17 2022 17:54:31
11:54:31.640 -> Hour: 17
````

I would expect the behaviour of Wokwi to match the real-world hardware's results. As @urish said in that comment thread, "My bet is that's some missing feature in the RTC".

yepher, 2022-01-17:
Printing the milliseconds in the sketch above appears to print the correct value.

```
struct tm* timeinfo;
 time_t now;
 time(&now);
 Serial.print("Time(now): ");
 Serial.println(now);
```

kartben, 2022-01-18:
@urish what version of the ESP Arduino core is Wokwi using? I think the culprit might be this recent change in 3.0.0 core? https://github.com/esp8266/Arduino/pull/7708. Could it be that the old version of newlib is still pulled somehow?

kartben, 2022-01-18:
Minimal repro case: https://wokwi.com/arduino/projects/321115858986336850

urish, 2022-01-18:
Thanks @kartben ! 

We're using Arduino ESP32 Core v2.0.2 (not the ESP8266 core). 

For reference, here's the complete list of cores we currently have on the build server:

```
ID             Installed Latest Name
arduino:avr    1.8.4     1.8.4  Arduino AVR Boards
attiny:avr     1.0.2     1.0.2  attiny
ATTinyCore:avr 1.5.2     1.5.2  ATTinyCore
esp32:esp32    2.0.2     2.0.2  esp32
rp2040:rp2040  1.9.11    1.9.11 Raspberry Pi Pico/RP2040
arduino:mbed   2.6.1     2.6.1  Arduino Mbed OS Boards
```

kartben, 2022-01-18:
Right, I'm not sure how I didn't realize I was referencing the ESP8266 core! That being said, the proiblem seems incredibly similar to what is being described here: https://forum.arduino.cc/t/ntp-server-doesnt-get-the-right-date-esp8266/938391

urish, 2022-01-18:
My observation so far: the issue is in `gmtime()`, whose implementation is coming from the ESP32 ROM. Investigating it now...

kartben, 2022-01-18:
Interestingly, the [SimpleTime.ino sample](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Time/SimpleTime/SimpleTime.ino) from the ESP32 core crashes. (see https://wokwi.com/arduino/projects/321121796856218194)

```
Stack smashing protect failure!

abort() was called at PC 0x400daecc on core 1

Backtrace:0x40083176:0x3ffb26a00x4008cd41:0x3ffb26c0 0x40091e65:0x3ffb26e0 0x400daecc:0x3ffb2760 0x400d33ce:0x3ffb2780 0x400d21f9:0x3ffb27f0 0x400d3c0e:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
```

urish, 2022-01-18:
Sounds like the same issue as #257. If this fixes both - great :)

urish, 2022-01-18:
Bug found - missing edge case in the implementation of the MULSH instruction (multiplies two 32-bit signed integers and returns the high part of the 64 bit result).

closed 2022-01-18 by urish

urish, 2022-01-18:
Should be fixed now :)

sutaburosu, 2022-01-18:
Awesome! Thanks, Uri.

urish, 2022-01-18:
Hooray!