wokwi/wokwi-features #278: spaces in library names cause issues

leftCoast, 2022-01-18:
In the properties file for libraries on the "relies on" line. if a library name has a space in it, it seems to screw up the parsing. 

Here is a sketch that has two libraries relying on one with space in name.
https://wokwi.com/arduino/projects/317539722329064002 

It just won't compile. 

**Environment **
- OS Mac
 - Browser chrome
 - Version Don't know. Not at my desk at the moment.

urish, 2022-01-18:
Thanks for reporting! 

Looking into this, it seems like there's some specific issue with the LC_neoPixel library - I tried to install it in the Arduino IDE, and it fails with an error:

```
 no protocol: 
java.lang.RuntimeException: java.net.MalformedURLException: no protocol: 
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:248)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.MalformedURLException: no protocol: 
	at java.net.URL.<init>(URL.java:593)
	at java.net.URL.<init>(URL.java:490)
	at java.net.URL.<init>(URL.java:439)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:64)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.libraries.LibraryInstaller.performInstall(LibraryInstaller.java:155)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:125)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:237)
	... 1 more
```

urish, 2022-01-18:
Looking at the [released version](https://github.com/leftCoast/LC_neoPixel/blob/4c3f6e1d1c1934fc4a6540d1ebd7a3124fc5c2d3/library.properties#L10), it seems like the `depends` list is missing a comma. 

This was fixed in the [master version](https://github.com/leftCoast/LC_neoPixel/blob/master/library.properties#L10). So I guess a new release is all we need?

leftCoast, 2022-01-18:
Damn! It ma fault again! Thanks I never would have caught that! Sorry to pester you with it.

closed 2022-01-18 by leftCoast