wokwi/wokwi-features #279: TM1637 Support possible?

RandoSY, 2022-01-19:
Would it be possible to implement TM1637 LED controller?
[image]
[image]

urish, 2022-01-19:
Thanks for the suggestion! I opened it for [voting](https://wokwi.com/features?sort=newest#feature-1108603271)

urish, 2022-08-09:
The TM1637 is now available in Wokwi.

- Documentation: https://docs.wokwi.com/parts/wokwi-tm1637-7segment
- Example: https://wokwi.com/projects/339227323398095442

closed 2022-08-09 by urish

erichuang2013, 2022-12-03:
As https://wokwi.com/projects/339227323398095442 us "tm1637.h", do we have this file in repo? or refer to any other project in github?

urish, 2022-12-03:
In the project you linked, the file comes from the Arduino library called "Grove 4-Digit Display" (see in the Library Manager tab). 

The sources for that Arduino library are hosted on GitHub: https://github.com/Seeed-Studio/Grove_4Digital_Display

RandoSY, 2023-10-13:
So glad to see this implemented. Is there an issue with Decimal point display?The TM1637 setup on the Open-Smart Rich UNO shield will do them OK, but I know there are issues with other modules ...

NewAbbreviations950, 2024-12-14:
Yeah I'm using the Pico SDK and the decimal points don't work. When sending the correct bytes to the tm1637 the bit corresponding to the dp segment doesn't seem to do anything.