wokwi/wokwi-features #280: ILI9341 Seems to drag down the simulation a LOT.

leftCoast, 2022-01-19:
**The issue**
ILI9341 Seems to drag down the simulation a LOT.

**You can see it here..**
When Drawing to the ILI9341 the simulation slows down and events are lost as it's spending much of its time seemly pushing pixels and not checking for "clicks" on the screen. Even trying to do simple animation drags it down really slow. To see this in action Open up the icon editor and try to do some "painting" Using a large brush. The % values drop typically to around 20% and updating dialog boxes can drop it down to around 10%.
[https://wokwi.com/arduino/projects/320027251914572370](url)

**Describe the solution you'd like**
I just want this to be Speedy Quick!

urish, 2022-01-20:
Did some performance hunting today. Does it work any faster now?

leftCoast, 2022-01-22:
It does seem faster. I can still see it refresh my pixels. (See that I'm doing a horrible job of optimizing the redraw.) :)

urish, 2022-01-23:
Great. So we picked the low hanging fruits and got some performance gains already.

Shell we keep this open (for voting) or close as this is "good enough"?

leftCoast, 2022-01-24:
Keep it open, its still kinda' painful for the icon editor.

leftCoast, 2022-04-04:
So I was playing with this today and it suddenly occurred to me. The slowdown I'm seeing now may NOT be the display, but the SD card. The example I gave is very SD card heavy in the code. The editing is done on the SD card not in RAM. SO there's a LOT of back and forth between the two.

urish, 2022-07-28:
Closing as this got no votes in the past 6 months.

closed 2022-07-28 by urish