wokwi/wokwi-features #282: Functional Reset button and input

guillaume-dorczynski, 2022-01-22:
Hello :)

**Is your feature request related to a problem? Please describe.**
There is no on-board Reset button, but it can be useful to press this button to reset the board instead of stopping/restarting the simulation. I've tried to connect a button to the Reset pin, but it doesn't work :(

**Describe the solution you'd like**
I would like at least the Reset pin to be functional, and better, if possible, add the on-board Reset button.

Thanks !

sutaburosu, 2022-01-22:
There's already [a feature request for this](https://github.com/wokwi/avr8js/issues/107) which is [open for voting](https://wokwi.com/features#feature-993565660). :)

guillaume-dorczynski, 2022-01-24:
Sorry, I had searched in the issues here but didn't find

I use Arduinos and ESP32

urish, 2022-01-27:
I guess we can look into this once #186 is in place. Right now, the microcontroller controls the timing for the entire simulation, so it's tricky to reset it without affecting the other parts.

urish, 2022-04-29:
Reset input implemented. The button is also functional (only for Nano, at the moment).

closed 2022-04-29 by urish

urish, 2022-06-07:
Uno / Mega now also have reset buttons