wokwi/wokwi-features #283: Sharing only the simulation, without the code

guillaume-dorczynski, 2022-01-22:
Hello :)

**Is your feature request related to a problem? Please describe.**
It would be very useful if it was possible to make the source code of a project visible only by its author, in a way that is impossible, or at least very difficult, for other users to retrieve the source code.

**Describe the solution you'd like**
I would like to have another link in the "Share" button so we can share only the simulation.

Thanks !

urish, 2022-01-22:
Hi Guillaume, thanks for opening this feature request!

Can you tell me more about the use case for this?

guillaume-dorczynski, 2022-01-24:
Hello Uri

Sometimes I get hired to write programs, and I would like to share the link of a simulation to the customers so they can try (at least partially, with what is possible to do in the simulator) before paying for the code :)

urish, 2022-01-24:
Thank you for explaining, that sounds like a very good use case. 

I opened this feature for voting: https://wokwi.com/features?sort=newest#feature-1111630848

guillaume-dorczynski, 2022-01-24:
This could also be useful for teachers, they could ask students to write the program to replicate the behavior of a simulation :)

urish, 2022-01-24:
A workaround for the time being:

1. Create a simulation with just the diagram (and empty code file that does nothing)
2. Send the ".hex" file to the customer
3. Ask them to open the link, click on the code editor, press "F1" and select "Load HEX file and Start simulation...", then open the ".hex" file you sent

guillaume-dorczynski, 2022-01-24:
Oh, I didn't know about F1! Thanks ;)

urish, 2022-07-28:
Closing, as this got no votes in the past 6 months.

closed 2022-07-28 by urish