wokwi/wokwi-features #284: Ability to specify the background colour of the diagram pane

sutaburosu, 2022-01-23:
When "saving a copy" of a project loaded with "?dark=1" in the URL, or when briefly invalid JSON is present whilst editing diagram.json, the background colour of the diagram pane reverts to the default bright background.

Could we please have the ability to specify in diagram.json a preferred background colour for the diagram pane? This may also help thumbnails to be generated in a more visually appealing way.

urish, 2022-01-23:
If I understand correctly, there are three different concerns here:

1. Personal experience - when in dark mode (through URL), there are glitches that happen when navigating between pages. Specifying the background color for the diagram won't fix the bright editor pane nor the bright flash when you have an invalid diagram.json. These need to be addressed separately. 
2. Thumbnails - #268 can be a workaround, but I agree it's more convenient to specify a background color in diagram.json and forget about it.
3. Experience for other users who open the project: if I understand correctly, we only want to specify the color of the diagram pane in this case (not the editor / other elements). Is there a need to support arbitrary colors? Or just dark/light as we have now?

sutaburosu, 2022-01-23:
1. The brief flashes when editing JSON aren't a big concern for me. A few times I have triggered a condition where the editor pane was dark, but the diagram pane was bright. I misremembered how to trigger that. After testing again, I think it was probably after seeing the "Invalid JSON in diagram.json. Press RETRY to try again."
2. :+1: 
3. Yes, just the diagram pane. A choice of colours would be ideal, so I could choose a black background for some projects.

leftCoast, 2022-01-24:
Just support a background image and that takes care of all of it. Colors, patterns, dark, light, patterns..