wokwi/wokwi-features #286: The <=100% cap on simulation speed depends upon screen refresh rate

sutaburosu, 2022-01-24:
When running on a display refreshing at >60Hz, the maximum speed of execution increases beyond 100%.

For example [this sketch](https://wokwi.com/arduino/projects/321708892771844691) runs at 99-100% at 60Hz. At 70Hz (116% of 60Hz) it rises to 113-116%. At 120Hz it 195-200%, and at 144Hz it gets to around 220% but is limited by my CPU speed from reaching the expecting 240%.

I can't test <60Hz refresh, but I suspect the simulation speed would be capped below 100%.

It would be nice if the maximum speed could be capped at 100% regardless of the rate of VSyncs.

 - OS: Linux (Kubuntu 21.10)
 - Browser: Chrome v97

urish, 2022-05-18:
Fix is up! Can you please test and report back?

sutaburosu, 2022-05-18:
That works great. Thanks, Uri.

closed 2022-05-18 by sutaburosu