wokwi/wokwi-features #287: Micropython simulator won't start.

apage-pi, 2022-01-25:
The Micropython (Pi Pico) simulation will not startup

https://user-images.githubusercontent.com/72623953/151030663-23a8000a-f83e-4cb7-a44b-a3faceb55016.mp4

.

urish, 2022-01-25:
Thanks for reporting!

Do other simulations work for you?

What browser/operating system are you using?

sutaburosu, 2022-01-26:
@apage-pi Spamming the start button isn't going to help. Depending on the speed of your computer, internet connection and what is cached in your browser, it can actually harm things.
It would also be interesting to know what browser extensions you are using, and whether it works without any extensions enabled. An easy way to test a site without any extensions is to use Incognito/Private browsing mode.

urish, 2022-03-01:
Closing as there is not enough information here to make this actionable.

closed 2022-03-01 by urish