wokwi/wokwi-features #291: ESP seems slow?

leftCoast, 2022-01-26:
**Describe the bug**
ESP 32 seems really slow.

**To Reproduce**
Here's a link : [https://wokwi.com/arduino/projects/321882926923907667](url) Just blinking an LED runs at around 20%.

**Expected behavior**
Blinking an LED should be easy peasy to run? More like 100%?

**Environment (please complete the following information):**
 - OS: Mac OSX
 - Browser chrome
 - Version 97.0.4692.99

**Additional context**
First time ever trying out an ESP-32 maybe I'm missing something? Also it got all weird about uploading a library. At first it couldn't see it after it was uploaded. Then I tried again and I got 2 libraries.txt tabs. In the end I had to delete one of them to get it working.

urish, 2022-01-26:
Workaround: adding `delay(10)` to the loop should speed it up a bit.

I haven't looked into optimizing the ESP32 simulation yet - focusing on getting it to work first :)

leftCoast, 2022-01-27:
Oh, ok. Just was kinda' surprised.

urish, 2022-01-31:
Quick update: did a bunch of optimizations to the code, so it should be noticeably faster now. You can use the new (optimized) version by adding "esp32.preview." in front of the URL, e.g.

https://esp32.preview.wokwi.com/arduino/projects/321882926923907667

The current version on wokwi.com still uses the unoptimized ESP32 simulation

leftCoast, 2022-01-31:
Interesting. Much faster. But it still wants the 4ms delay in there. Supposed deeps off as lot without it.

urish, 2022-01-31:
What speed difference do you observe?

leftCoast, 2022-01-31:
with delay 99-100
without 60-70s

iPhone

urish, 2022-01-31:
Thanks! And how does that compare to the old, optimized version?

leftCoast, 2022-01-31:
Chrome original
with delay - 80s
no delay - 15

Chrome new stuff
with delay - 99-100
no delay - 30s

iPhone original
with delay - low 70s
no delay - 20

iPhone new stuff
with delay - 99-100
no delay - 70s

note on iPhone - old version the keyboard edit thing is hidden below scrolling distance.
new version, its further up and you can see and click it.

urish, 2022-02-27:
Should we keep this issue open?

leftCoast, 2022-02-28:
I don't know. It's still 5-10% slower on the phone with no delay (). You ready to call that good enough?

urish, 2022-02-28:
It's getting to the point where it's becoming increasingly hard to optimize - as we've eaten all the low hanging fruits

leftCoast, 2022-03-01:
I wonder if it needs a minimum of NOPs to function? that's what it feels like. Maybe some long program would have enough without the delay?

urish, 2022-03-01:
It goes like this: when you have a busy loop (one without any kind of delay), the CPU has keeps executing instructions. So we need to simulate these instructions, and this is computationally heavy. There's no easy way to know what the simulated code is doing, so we just need to keep running the simulation.

When you use `delay()` in the ESP32/Raspberry Pi Pico, the code doesn't do NOPS - it rather tells the CPU to go to sleep. 

So if your loops has a 9ms delay then code some that takes ~1ms to run (e.g. reading a button, checking for WiFi packets, etc), we actually skip 90% of the work by fast-forwarding these 9ms. There are also interrupts that we need to serve, so it's somewhat less than 90% of the work that we skip, but still - it saves the simulator a lot of computation.

Not sure why it's faster as more code added - does it seem significantly faster?

closed 2022-06-14 by urish