wokwi/wokwi-features #293: Project reappearing after being deleted

guillaume-dorczynski, 2022-01-28:
Hello :)

**Describe the bug**
When I try to delete a project in My Projects page, the project disappears from the list, but reappears less than a second later (or sometimes when I refresh the page), I have to delete it one more time..!

**Expected behavior**
Of course it should be deleted once and for all

**Environment (please complete the following information):**
 - OS : Windows 10
 - Browsers : Chrome, Edge, Firefox
 - Version : Latest

sutaburosu, 2022-01-29:
I have noticed this too, but my understanding is that this is a temporary thing caused by caching. In my testing, if I delete a project and then reload the projects page twice, the deleted project no longer appears. Does this work for you too?

urish, 2023-11-13:
Thanks for reporting, fixed!

closed 2023-11-13 by urish