wokwi/wokwi-features #295: SSD1306 on ESP32 - mirrored display?

kartben, 2022-01-31:
**Describe the bug**
It would appear as if SSD1306 display is reversed on ESP32

[image]

**To Reproduce**

https://wokwi.com/arduino/projects/322316932640408148

urish, 2022-01-31:
Looks like our SSD1306 implementation is missing the segment remapping feature:

[image]

(from the [datasheet](https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf))

sutaburosu, 2022-01-31:
Another difference between this [beta library](https://github.com/ewertons/esp32-azureiotkit-sensors/blob/main/src/ssd1306.c#L268) and [Adafruit's SSD1306 library](https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.h#L88-L89) is the row scan direction.

urish, 2022-01-31:
Thanks @sutaburosu, I think you nailed it down. It wasn't the segment remapping, rather the row scan direction. Fix is up!

closed 2022-01-31 by urish

kartben, 2022-01-31:
Wow that was fast. I can confirm the fix works (although not on the esp32.preview server yet, right?)

urish, 2022-01-31:
Right, that one should update in roughly 5 minutes

kartben, 2022-01-31:
aaand it works on the preview server too. Thanks again!

v1d32, 2024-12-07:
I need mirrored how to make normal to mirrored?