wokwi/wokwi-features #296: MPU6050 does't work properly on ESP32

kartben, 2022-01-31:
**Describe the bug**

Running virtually any of the basic MPU6050 samples on ESP32 (using Adafruit-MPU6050) seems to show that the sensor is not properly working on the platform. Only accel_x seems to work

[image]

**Additional context**

I have another app that is not using the Adafruit lib and that is also not behaving properly and it hangs when trying to initialize the sensor.

urish, 2022-01-31:
Thanks for reporting! Can you please share a link to a project that reproduces the issue?

kartben, 2022-01-31:
Sorry, I thought I had -- I have just updated the original bug desc

kartben, 2022-01-31:
Maybe related: although the "loop()" in the repro project is only waiting for 10ms, and the simulation running at close to 100% speed, the traces on the serial console get updated only every 1sec or so, which is not the case using the exact same code in  an Arduino Uno AVR sim

urish, 2022-01-31:
Turned out to be a bug in the ESP32 I2C Implementation. It wouldn't allow you to read more than 2 bytes in one go. 

Can you please test it again (against the esp32.preview branch)?

urish, 2022-02-01:
Note: the preview branch has just been merged, so the fix should also be present on wowki.com

kartben, 2022-02-01:
That's much better, yes! Any idea about why the sensor reading is getting refreshed only every 1sec or so? Is there something with I2C on ESP32 being slower? This sample seems to demonstrate that I2C writes are indeed really slow: https://wokwi.com/arduino/projects/322380195661087314

kartben, 2022-02-01:
I just spotted another issue, that can be seen by running the same Adafruit lib sample as earlier (https://esp32.preview.wokwi.com/arduino/projects/322322941068968530):

```  mpu.setAccelerometerRange(MPU6050_RANGE_8_G);

this seems to not work properly, as "getting" the acceloremeter range right after returns 0 aka MPU6050_RANGE_2_G. This works fine in the AVR sim

urish, 2022-02-01:
> This sample seems to demonstrate that I2C writes are indeed really slow:

The best way to look into it is hooking a [Logic Analyzer](https://docs.wokwi.com/guides/logic-analyzer).  Then you can actually see the time spend transmitting data over I2C.

In the first, faster example:

[image]

You can see each burst takes ~7ms.  Adding the 10 ms delay in your loop, you get about 17 milliseconds between updates (or about 60FPS). Also, the simulator is sleeping more than 50% of the time (in the delay loop), hence it can run the simulation faster.

However, looking at the slower example you'll see much more time is spent on the I2C communication:

[image]

Here, each burst takes around ~60ms. So from the simulated time perspective, you should get a frame every 60 milliseconds (so around 16 FPS). However, this also means that simulator is busy most of the time (60 out of every 70 milliseconds), so the simulation runs slower and your actual FPS is way slower. The funny thing - if you bump the delay to let's say 100, the simulator speeds up and you still get the same result.

As far as I can tell, the reason for the slow down is that the ESP32 Arduino implementation busy waits for the I2C transmission to complete, so as long as data is transmitting, the simulated ESP32 is just busy burning CPU cycles. Unfortunately, there's not much we can do here in terms of performance.

urish, 2022-02-01:
> I just spotted another issue, that can be seen by running the same Adafruit lib sample as earlier (https://esp32.preview.wokwi.com/arduino/projects/322322941068968530):

Reproduces, seems like some of the I2C writes are not going to the MPU-6050. Needs further investigation.

urish, 2022-02-01:
After several hours of tedious debugging, I finally found the culprit: the acks for I2C read operations were inversed.

The datasheet says:

[image]

Which I interpreted: send ACK when ack_value is high, and NACK when ack_value is low. It turns out it works the opposite way. What they probably wanted to say is that the ack_value should be copied to the SDA line (I2C pulls it low when it wants to send an ACK, and leaves it high when it wants to send a NACK).

The MPU6050 library read from the register before updating it. As the read failed, it silently skipped the write.

Fix should be up on wokwi.com in a few minutes. 

Note: you can stop using the esp32.preview branch, as it has been merged and now available on wokwi.com. This branch is no longer receiving updates or bug fixes.

kartben, 2022-02-01:
Super cool - thank you so much! It now works fine for me, and I've learnt a lot about the logic analyzer in the meanwhile. REALLY cool stuff :)

Thanks for the note/update regarding the now dead esp32.preview branch, too!

closed 2022-02-01 by kartben

urish, 2022-02-01:
Thank you for finding all those hidden bugs :)