wokwi/wokwi-features #297: Deleting a file tab doesn't delete.

jflamy, 2022-01-31:
**Describe the bug**
In the code editor, deleting a tab does not remove it.

**To Reproduce**
Open a project, delete a tab.
Go away.
Reopen wokwi, go back to the project, the tab is back.
could be related to #293 

**Expected behavior**
Tabs should go away.

**Environment (please complete the following information):**
 - OS: Windows 11
 - Browser Latest Chrome

**Additional context**
It would be nice to be able to mark tabs as "private". Things like keys and certificates could be hidden when sharing the project to ask for help.

sutaburosu, 2022-01-31:
I can't reproduce this. Are you saving the project after deleting a tab? You may have to make some other change (add a blank line) to trigger the ability to save.

jflamy, 2022-01-31:
Given that there is an explicit feedback ("This will remove tab xyx do yo agree") which is normally used when an action takes place immediately and cannot be undone, and that the tab actually disappears AND that the save button does not get highlighted, this is not intuitive.
If removing a Tab requires a Save, then the Save button should become available.

urish, 2022-01-31:
> If removing a Tab requires a Save, then the Save button should become available.

Agreed

jflamy, 2022-01-31:
Also, renaming a Tab does cause "save" to become active.

urish, 2022-02-09:
Fixed!

closed 2022-02-09 by urish