wokwi/wokwi-features #299: Addition of a GPS Module

TheAlexBas, 2022-01-31:
**Is your feature request related to a problem? Please describe.**
I want to plan out a project that requires GPS. The problem is that Wokwi currently does not have any GPS modules.

**Describe the solution you'd like**
Implementing a GPS module to Wowki. I would assume the GPS output would either be blank or random.

urish, 2022-01-31:
Thanks for the feature request! Is there a specific module (or type of module) you'd like to see added? Or should we just pick the most popular one?

TheAlexBas, 2022-01-31:
The most popular one seems like a good idea. Thanks for the fast reply!

urish, 2022-01-31:
Now we just need to figure out which one is most popular. Then we can open this for voting!

TheAlexBas, 2022-01-31:
From my own research + a member of the Discord led me to [this](https://www.u-blox.com/en/product/neo-6-series). I guess the cheapest one from that selection would be the best idea (for my wallet lol) to vote on.

urish, 2022-01-31:
- Relevant tutorial: https://create.arduino.cc/projecthub/ruchir1674/how-to-interface-gps-module-neo-6m-with-arduino-8f90ad
- Arduino library: https://github.com/neosarchizo/TinyGPS/

Opened for [voting](https://wokwi.com/features?sort=newest#feature-1119915232)!

leftCoast, 2022-02-01:
Can I vote for this one?
[https://www.adafruit.com/product/746](url)

urish, 2022-02-01:
@TheAlexBas any objections that we switch to the one @leftCoast suggested?

TheAlexBas, 2022-02-01:
No objections!

urish, 2022-02-01:
Updated

urish, 2022-10-01:
Closing as this got no votes in the past 6 month

closed 2022-10-01 by urish

HabibMCS, 2023-05-12:
@urish    There is always a community interested, even if no votes. This seems to be a useful tool so it should be available. I am up for this.

urish, 2023-05-14:
@HabibMCS thanks for pointing this out. You can try creating a simulation model for the GPS module using the Chips API and share it with the community. 

Here's the documentation for the Wokwi Chips API: https://docs.wokwi.com/chips-api/getting-started