wokwi/wokwi-features #300: DS1307 initial conditions

alto777, 2022-02-01:
The sim DS1307

"....is automatically initialized to the current system time when starting the simulation. It then keeps counting the time."

Add an attribute to the part to let it power up just like a real DS1307. 

I can't say exactly what would be in the registers. ISTR that the date and time are not even plausible, but they certainly aren't magically the current date and time. The data sheet may say, otherwise we get empirical and see by looking. IRL.

This fact must be accounted for by ppl who use the part or module IRL. Therefor we should be able to simulate that when we need to create or fix code to work with every DS1307 that will be attached.

a7

urish, 2022-02-01:
Thanks for the feedback, it does make sense. Once caveat, though, unlike the real DS1307, the simulator does not remember that state of the DS1307 between executions, so you'll have to set the date every time (unless we change the implementation)

alto777, 2022-02-01:
Ironically it was trying to get the RTC to forget the time it was keeping that led me to the documentation.

I thought it was being cleverly cached, I tried everything short of re-installing the operating system <-- _hyperbole_ to get it to "malfunction" correctly.

So yeah, ideally there'd be some means of simulating having / not having a battery on the module. That sounds like a multiple nightmare.

I think just letting it power up random (by attributed choice) would suffice. Perhaps if that is easy, do it and wait for the even more ~~insane~~ ~~fussy~~ _meticulous_, that's the ticket, will raise the issue again.

a7

M-square, 2022-06-02:
This is a simulator - not RealLife ;-)

Slighty less ambitious feature request :
attribute = { "startDateTime": "220602160700", "start24h": "true" }
i.e. you can set a fixed value from which it starts. This sort of simulates it remembers the previous value or a pseudorandom value (which is the same everytime - https://xkcd.com/221/ )

urish, 2023-07-28:
Added an "initTime" attribute that allows you to set the initial time for the DS1307: https://docs.wokwi.com/parts/wokwi-ds1307#attributes

closed 2023-07-28 by urish