wokwi/wokwi-features #303: PCD8544/Nokia-5110 LCD Screen

Kalixtan, 2022-02-03:
These screens are a classic when it comes to DIY/electronics community's, these monochrome LCDs most commonly found on old Nokia phones but are very useful for small displays for low cost projects,

It would be lovely to have this classic bit of hardware added into Wokwi.

**[Datasheet](https://www.sparkfun.com/datasheets/LCD/Monochrome/Nokia5110.pdf)**

urish, 2022-02-04:
Thanks!

I opened this request [for voting](https://wokwi.com/features?sort=newest#feature-1123665318).

Implementation details: Uses SPI; Arduino library: https://github.com/carlosefr/pcd8544

urish, 2025-01-15:
Custom chip implementation: https://wokwi.com/projects/420132927195935745

urish, 2025-02-11:
Now also implemented natively in Wokwi:

- https://docs.wokwi.com/parts/wokwi-nokia-5110-screen
- https://wokwi.com/projects/422590948073922561