wokwi/wokwi-features #305: Simulation speed control

drf5n, 2022-02-08:
**Is your feature request related to a problem? Please describe.**
 It appears that the animation of the LED is limited in speed, making it difficult to see activity at less than 100ms or so.  You can see this in https://wokwi.com/arduino/libraries/demo/blink if you shorten the time to 100ms.  If you hit pause, the animation appears to be caught in an intermediate state.  I'm not worried about the animation of the LED, but that you can't use the simulation to examine fast processes.  

**Describe the solution you'd like**
I would like an option/slider/dial/tool to run the simulator at lower speed to accurately capture behavior that I cannot detect at full speed. 

If you could toggle the simulation speed control with a pin (i.e, control full/slow speed with a logic input), that would be awesome.

    // critical area
    digitalWrite(SlowSpeedPin,1);  // Config Wokwi/diagram/SimSpeedControl for 100Hz+pinEnabled  
    for(ii = 1;ii<100; ii++)
    {
        SoftwareSerial2.write(0xF5);
        digitalWrite(LED_BUILTIN,!digitalRead(LED_BUILTIN));
    }
    digitalWrite(slowSpeedPin,0);
    // run as normal

drf5n, 2022-02-08:
I do see that there were https://github.com/wokwi/wokwi-features/issues/236 and a sketch demonstrating the led persistence:  https://wokwi.com/arduino/projects/317535907128279616

I'm more interested in seeing a slow simulation speed for troubleshooting code.

urish, 2022-02-09:
Thanks for sharing your feedback!

I see that you already read through the existing issues, which describe the challenges of simulating LEDs and PoV.

Have you tried the [logic analyzer](https://docs.wokwi.com/guides/logic-analyzer)?

It lets you look at the signals in a nanosecond level accuracy, so it may be a good fit when troubleshooting code?

You can see an example of me using it to figure out a bug with LED timing in [this video](https://www.youtube.com/watch?v=YUUHw-bU9yk), starting at minute 5:18

closed 2023-07-23 by urish