wokwi/wokwi-features #306: Simulated USB with monitoring

jflamy, 2022-02-09:
When emulating a USB keyboard, it would be nice to have a monitor of some sort to see the key presses being sent.

Conversely,  a virtual keyboard to send keypresses (and perhaps mouse interaction) would be nice.

kartben, 2022-02-09:
Or maybe similar to wokwi-gdbserver, perhaps there could be a wokwi-usbipserver which exposes the device's USB port over USB-IP? It would then be reasonably straightforward to mount it on the host machine. See e.g. https://docs.microsoft.com/en-us/windows/wsl/connect-usb

jflamy, 2022-02-09:
@kartben yes, some form of USB proxy, with the ability to peek what goes in and out.

urish, 2022-02-09:
Than @kartben! As far as I can tell, usbipd-win does the opposite: it will enable sharing of Windows USB devices with whatever is running in the simulator, in case we implemented USB Host support. 

By we need here the other end of the USB-over-IP connection: use a shared USB device in windows. Do you know if such thing exists (and is maintained)?

jflamy, 2022-02-09:
Have not thought this fully, but would taking the usb output from the
simulator and feed it (somehow) to a USB over IP gateway work ?
The simulator could perhaps emit json-formatted USB "packets" and the
gateway would convert that  and become the client to a Windows USBIP server.
The other way might work as well (gateway implements server side and feeds
the simulated USB)

https://github.com/cezanne/usbip-win

ftab, 2022-02-11:
There's also WebUSB API

https://wicg.github.io/webusb/

urish, 2022-02-15:
From my experience with WebHID, a few months ago, it's very limited. It doesn't let you use any HID device, and specifically excludes mice, keyboards, etc. You can learn more [here](https://github.com/robatwilliams/awesome-webhid#good-to-know)