wokwi/wokwi-features #309: Same code runs fine on ESP32, extremely slow on ESP32-S2

jflamy, 2022-02-11:
**Describe the bug**
The same code and the same circuit runs extremely slowly on ESP32-S2

**To Reproduce**
See the following two projects
[Working ESP32 project](https://wokwi.com/arduino/projects/322534543008465491)
[Extremely slow ESP32-S2 with same code](https://wokwi.com/arduino/projects/323248259320513108)

S2 Simulation startup is extremely slow, and never gets to the point where serial output is shown.  CPU Usage is high.

For S2 I used https://www.embarcados.com.br/franzininho-wifi-entrada-analogica-esp-idf/ as reference for the pinout, and I cannot see any obvious error in the wiring (all grounds are connected together)

**Expected behavior**
Similar behaviour between the two variants.

**Environment (please complete the following information):**
 - OS: Windows 11
 - Browser Edge
 - Version 98.0.1108.50 64 bits.

**Additional context**
None.

ftab, 2022-02-11:
I tried this on an iPhone 12 mini on iOS 15.3 and the S2 is slightly slower (50-70% simulation speed vs. mostly full speed for the ESP32) but does seem to boot up and start. It just misses the "Connecting to WiFi Wokwi-GUEST" and the first thing the serial terminal sees is some dots.

Wild that this stuff even works on a phone...

jflamy, 2022-02-11:
I the slow behaviour is actually on esp32.preview.wokwi.com.   I also observe strange behavior with respect to timing and buttons (in addition to the first Serial output being lost).  I'm getting the gist of using the logic analyser, and I will be opening separate issues.

urish, 2022-02-12:
"esp32.preview.wokwi.com" is not up to date, it's running an older version, without the recent performance optimizations and bug fixes. Please do not use it.

Also, when using Serial on the ESP32-S2, it's advised to add the following line at the beginning of `setup()`:

```cpp
while (!Serial) delay(50);
```

This ensures that the USB-CDC link is established before you write any output to the Serial console. Otherwise, you may miss the first few prints.

Also, it seems like your could might use TLS - note that it's not yet supported on the ESP32-S2 (we are missing the crypto hardware accelerator). I will keep updating on the progress of the TLS implementation in #196 .

closed 2022-02-12 by urish