wokwi/wokwi-features #311: Make a non-Arduino simulator for the RPi Pico

eleciawhite, 2022-02-14:
I want to use Wokwi for a remote-classroom. However, the Arduino IDE is one I'm trying to get the students away from.

Can you make an RPi Pico Project that is C/C++ based? It would still hook to the [Pico SDK](https://raspberrypi.github.io/pico-sdk-doxygen/modules.html) as it does now. Instead of setup() and loop(), the new Project type would show a main function like:

```
#include "pico/stdlib.h"

const uint LED_PIN = 2;

int main() {
  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);
  while (true) {
    gpio_put(LED_PIN, 1);
    sleep_ms(250);
    gpio_put(LED_PIN, 0);
    sleep_ms(250);
  }
}
```
I want to be able to give students PWM code for an LED and have them modify it to hear a buzzer in driven by PWM. I also want them to be able to construct final projects (your peripheral list is great!).

urish, 2022-02-26:
First working prototype: https://wokwi.com/projects/new/pi-pico-sdk

It compiles against Pi Pico SDK 1.3.0. Only "main.c" is currently supported - you can add additional .h files and it'll probably work, but additional ".c" files won't get compiled due.

@arcostasi Thanks for the comment! I did some testing and figured out that the Arduino core is missing some of the basic functionality (e.g. `stdio_init_all()` is missing), so now we're going to have native Pi Pico SDK builds.

urish, 2022-03-05:
Good news - multiple .c/.cpp source file are now supported!

Here's @arcostasi's example without the "include" workaround: https://wokwi.com/projects/325323976447361620

And our beloved blink: https://wokwi.com/projects/325324198971966034

closed 2022-03-05 by urish