wokwi/wokwi-features #312: ESP32-S2 Declaring a variable causes digitalWrite() not to happen (!)

jflamy, 2022-02-15:
**Describe the bug**
Adding a declaration cause digitalWrite() to fail

**To Reproduce**

1. Open https://wokwi.com/arduino/projects/323622229326692947
2. Run the simulation. Observe the blinking LED
3. Uncomment line 4.
4. Run the simulation. Observe that the LED no longer blinks.

**Expected behavior**
LEDs should blink.  Not a C++ guru, but seems that something strange is happening in the constructor.  Yet these two other projects using a USB HID library do not exhibit this strange behaviour even when all the code is kept.

- [esp32-hal-tinyusb](https://wokwi.com/arduino/projects/323245136964223572)
- [subset of ESPTinyUSB](https://wokwi.com/arduino/projects/323337115942257236)

**Environment (please complete the following information):**
 - OS: Windows
 - Browser: Edge
 - Version 98.0.1108.50 64-bit

**Additional context**
Add any other context about the problem here.

urish, 2022-02-15:
Thanks for the detailed report!

As far as I can tell there's an issue with the logic of the program:

```cpp
  if (toggle) {
    digitalWrite(11, LOW);
    toggle = false;
  }
```

When you first enter the loop, you set `toggle` to false, and then it'll never get into this `if` statement again, effectively turning `loop()` into doing nothing (just a bunch of `delay(10);`).

If you change the logic to read:

```cpp
  digitalWrite(11, toggle ? HIGH : LOW);
  toggle = !toggle;
```

Then it keeps toggling the pin:

https://wokwi.com/arduino/projects/323655439495463506

jflamy, 2022-02-15:
The toggling Is _not_ the issue.  The issue is that if you **uncomment line 4 there is no LED at all.**

urish, 2022-02-15:
Should be fixed now - can you please check again?

jflamy, 2022-02-15:
The other two libraries don’t have that issue which is why I opened the
ticket.  I was surprised that this one behaved that way.

-- 
(Envoyé de mon téléphone mobile.)

urish, 2022-02-15:
It was a missing register in the USB peripheral - the library waited for the USB reset to be complete, but because the register wasn't implemented, it kept waiting forever...

closed 2022-02-15 by jflamy