wokwi/wokwi-features #314: Can we please get support for ESP8266

wavesailor, 2022-02-22:
Please could we get support for the ESP8266. 
I know it is older but there are still so many of them being used.

https://en.wikipedia.org/wiki/ESP8266

urish, 2022-02-22:
Thanks for the suggestion! What would you use the ESP8266 for?

wavesailor, 2022-02-22:
Firstly I'd like to try build all those weather stations with the ESP8266. 
This website has this one: https://randomnerdtutorials.com/esp8266-dht11dht22-temperature-and-humidity-web-server-with-arduino-ide/ 
I then like to add a battery to it if possible.

urish, 2022-02-22:
Thanks! I opened this request [for voting](https://wokwi.com/features?sort=newest#feature-1147210537)

juliocr1, 2022-08-31:
The use of ESP8266 is very vast, we have many projects distributed, this simulator would be a very important resource.

urish, 2022-08-31:
> we have many projects distributed

Who is "we"? What projects do you have?

urish, 2022-09-01:
Thanks for sharing your thoughts @juliocr1!

donald2407, 2022-12-16:
**Please get support for the ESP8266.**

The3ven, 2023-02-19:
I'm upvoting for the esp8266 board too because I need to learn how this thing works for my FTP server

Ricardo-SS, 2023-03-15:
Adding ESP8266 would be good for me too. It's very cheap and easy to use. I am also using this mcu.

infrafast, 2023-05-23:
I would like to have the simulator as well.

urish, 2023-06-25:
Update: we open sourced a basic ESP8266 simulation library

https://github.com/wokwi/esp8266js

hanpal, 2023-11-30:
> Thanks for the suggestion! What would you use the ESP8266 for?

I use it for all my projects since it's much cheaper than the ESP32. I have programmed about 50 ESP8266 using Wemos D1 mini or pro for home automation projects and have never even considered using ESP32 since the much cheaper ESP8266 has always been more than sufficient. And it has NOT been replaced by the ESP32.

Phetdanai, 2023-12-22:
Thanks to @urish for that ".zip" file.
Download Links:
[esp8266](https://github.com/wokwi/wokwi-features/files/13748628/esp8266js-main.zip)

Laurent6012, 2024-03-25:
Hello,
And to be honest, like my father said, it's not usefull to use a hammer to kill a fly.
ESP8266 have alreday access to Wifi, enough IO for basical current and simply tasks.
**the IOT's world isn't only for rich people, not be ?**
So, using an ESP32 to manage only 1 sonde of temperature and humidity and even sending the result, my god, it's the perfect way to really becoming poor soon, i laugh.
No matter, have a good day

Ricardo-SS, 2024-06-25:
Considering the prices prevailing on the market today and the availability of several esp32 models, I think it will no longer be an advantage to develop for the esp8266, I am migrating all my projects to the esp32 as the prices are almost the same. I think it's a waste of resources to use an esp32 in some projects, but as they are almost the same value, it's better to have the resources that are missing.

gc2ceo, 2024-09-29:
I'm seconding ESP8266 simulation since I got some ESP8266 boards in real life.

FelixHo97, 2025-06-24:
> Update: we open sourced a basic ESP8266 simulation library
> 
> https://github.com/wokwi/esp8266js

Hi urish, Can this version be directly connected to WiFi? Does this open source project have a corresponding instance? For example, lighting up LED lights

urish, 2025-06-24:
Nope, there is no simulation model for the ESP8266 WIFI. Building one would require reverse engineering the WiFi stack first, similar to what I have done with the ESP32 a few years ago: https://www.youtube.com/watch?v=XmaT8bMssyQ