wokwi/wokwi-features #317: Wokwi ESP32 Simulator

takatik, 2022-02-28:
Hi,
I am a student and I’m tying to connect to an MQTT broker with ESP32 using the online simulator Wokwi but it doesn’t work (failed with state -2).

[https://wokwi.com/arduino/projects/316434364352168512](url)

Is there a problem in the code? Or do I need to be a member of the Wokwi Club to make it work?
Any help will be greatly appreciated.
Thank you

urish, 2022-03-01:
There are sometimes issues with the Public Wokwi IoT Gateway, otherwise it should work. Members of the Wokwi Club can [run the IoT gateway locally](https://docs.wokwi.com/guides/esp32-wifi#the-private-gateway), so it runs faster and more reliably.

I restarted the Public gateway now, can you try again and report back?

TakHamza, 2022-03-02:
Yeeeeeees! It Works 
Great Great Great a big thank you! It works perfectly

takatik, 2022-03-02:
Yeeeeeees! It Works 
Great Great Great a big thank you! It works perfectly

Biiiig Thaaaanks for your help