wokwi/wokwi-features #318: Rust as C/C++ alternative

Chris2011, 2022-03-01:


urish, 2022-03-01:
Hi Chris!

There's an initial prototype for Rust + ESP32 on Wokwi:

* [Hello World](https://wokwi.com/arduino/projects/323155109094621778)
* [Blinky](https://wokwi.com/arduino/projects/323845300685898322)
* [LED PWM](https://wokwi.com/arduino/projects/323765044769718868)

These are still early days, and I'm working together with the team at Espressif to improve the experience.

Rust + Arduino is not on the roadmap for now.

I'm curious, how did you find out about the simulator?

Chris2011, 2022-03-02:
Hey Urish,

thx for the fast reply. I'm sry I must have overseen it. Looks great. :).

I just googled for arduino simulator and I opened a lot of links and I think you have a direct organic search result at google and I think I found a website with some links and you were linked there.

I would like to say a big thank you for your great work and I'm looking forward what comes next :)

Chris2011, 2022-03-02:
And just now I found an article at heise: https://www.heise.de/news/Online-Simulator-fuer-Arduino-Raspi-Pico-und-ESP32-6324290.html unfortunately it is in german. It is just a little review about your nice tool :)

closed 2022-03-02 by Chris2011