wokwi/wokwi-features #319: BDF font cannot be uploaded

muranyia, 2022-03-01:
**Describe the bug**
The web UI doesn't allow uploading a .bdf font file which would be used with the (otherwise functional) `adafruit_bitmap_font` library.

**To Reproduce**
1. In the editor, choose "Upload file(s)..."
2. Select a .bdf file

**Expected behavior**
The file would be uploaded

**Environment (please complete the following information):**
 - OS: [Linux]
 - Browser [Firefox]
 - Version [97.0]

**Additional context**
Adafruit's CircuitPython font tutorial is [here](https://learn.adafruit.com/custom-fonts-for-pyportal-circuitpython-display/bitmap_font-library).

urish, 2022-03-01:
Can you try to change the file extension to .txt and upload it and see if it works?

muranyia, 2022-03-02:
It seems to work as I get `ValueError: Unknown magic number b'STAR'` now. I cannot test elsewhere at the moment as I'm waiting for my board to arrive and I cannot run this on the pc.

urish, 2022-03-02:
Thanks for joining the Wokwi club!

Can you share the link to the current version of your project? I believe there's an easy workaround, perhaps renaming the .txt file to .bdf as soon as the simulation starts (using `os.rename()`) will do the trick.

muranyia, 2022-03-02:
[Here](https://wokwi.com/arduino/projects/324894251813962324) it is. I can try only later today.

urish, 2022-03-02:
Thanks! It required a few more tricks, but I think I managed to get it to work:

https://wokwi.com/arduino/projects/325020559538651730

Note the addition of boot.py which makes the file system writable and renames the .txt file to .bdf

muranyia, 2022-03-02:
Great, thank you very much! I'm wondering though what was the issue with the txt... I might go back and try to find out.

As for the original issue definition, do you think .bdf will be added to the list of allowed files types?

urish, 2022-03-04:
> do you think .bdf will be added to the list of allowed files types?

More likely we'll find a general solution for uploading any kind of file. It's sort of already there right now: you can upload the file to an SD Card and connect it to the simulated Pi Pico, then copy the file off the SD Card, but this is as far as I can imagine from being a straightforward solution...

urish, 2022-03-26:
Closing this issue for now as there's a workaround.

closed 2022-03-26 by urish